miguelbalboa/rfid #324: version v1.3.6 not working with ESP32 and NodeMCU-esp8286 12E

akeilox, 2017-06-14:
### Step 2: Describe your environment

 * OS version: Windows 10 x64
 * Arduino IDE version: 1.8.3
 * MFRC522 Library version: 1.3.6
 * Arduino device: esp32 and nodeMCU-esp8286-12E
 * MFRC522 device: version 2.0
  
### Step 3: Describe the problem
If I use the 1.3.6 release version both esp32 and esp8286-12E does not work and hangs on boot after init.
If I manually switch back to version 1.3.5 then it works immediately on both esp32 and nodemcu 12e
#### Affected file(s) or example(s):
  
  
#### Steps to reproduce:

  1. on latest version of arduino ide in windows 10, add mfrc522 lib which defaults to latest version 1.3.56
  2. use the mfrc522 examples for esp32 and nodemcu esp8286 
  3. on boot and init, esp32 throw exception and hand and esp8286-12e nodemcu will act like no card detected
  
#### Observed Results:

  had to find the hard way after trying different pins, different units and different code examples, its simply the 1.,3.6 causing the issue but 1.3.5 works off the bat.
  
#### Expected Results:

  latest version to work with both nodemcu-esp8286-12e and esp32
and had to manually fall back to ealier 1.3.5 version.
  
#### Relevant Code:

any example code for mfrc gives the same behavior described above

Rotzbua, 2017-06-14:
Interesting, is there no compile error? The only change is 8db7e2ba4f1c6a791f0fcfcef470a3aa76adf3af , so no functional change.

akeilox, 2017-06-14:
yes no compile error, which made me try all other possible combinations like trying a port of the mfrc522 in c# with windows iot which worked, arduino uno worked, but not the esp8286-12e (nodemcu) and esp32.
tried pin combinations and all sorts examples to see where the issue might be. 
I didnt understand what could be wrong after changing 4 mrfc522 device, cables, and various working examples reportedly....

till i came across a post on the video https://www.youtube.com/watch?v=waYM37_fUsg
with comment from Syama Mishra.

After switching to 1.3.5, both esp32 and the older esp8286-12E worked without a hitch with no code changes.

lhcavalcanti, 2017-06-17:
I had the same issue this week, since monday I have been trying using the RFID sensor at the Node MCU version 1.0. I tried 2 nodes and 1 esp-12E, nothing worked.

**Step 1: Describe your environment**

OS version: macOS Sierra 10.12.5
Arduino IDE version: 1.8.1
MFRC522 Library version: 1.3.6 (but download from master)
Arduino device: esp32-12E and nodeMCU-esp8286-12E

**Step 2: Describe the problem**

If I use the 1.3.6 version, both nodeMCU and esp-12E, does not work and do not read any rfid card.
After days trying every configuration of pins, my friends manually switch back to version 1.3.5 (on the Arduino IDE library manager) then it works immediately.

**Reproducing the problem:**
1. Download library from github. (directly on the master).
2. Install the library on Arduino IDE.
3. Connect the to the Arduino the RC522 on the correct pins.
4. Try to read any data.

I have tried the example rfid_read_personal_data, and with the library from github nothing were read. As I had downloaded the library from github master, may the commit https://github.com/miguelbalboa/rfid/commit/8db7e2ba4f1c6a791f0fcfcef470a3aa76adf3af isn't the problem. 

Looking on every commit after 1.3.5 update, I suspect from:
https://github.com/miguelbalboa/rfid/commit/f07049c24e80f69fa4c1a2a0516c1fcff92df69b
https://github.com/miguelbalboa/rfid/commit/9c221e59ee962b1df65b439e0c5604e86766bbd6

Rotzbua, 2017-06-19:
> 1.3.6 (but download from master)

Master is not a stable branch, it is more a development branch. So please use a real release from https://github.com/miguelbalboa/rfid/releases .

Both commit you mentioned are not part of 1.3.6, they will be part of 1.4.x.

5chufti, 2017-08-29:
Hi,
I'm using the arduino 1.8.1 IDE portable on W7/32 with manual integration of esp8266 support (update_sdk_2.1.0 branch from github) and rfid (master from github) and "DumpInfo" example compiles and works without any problem on NodeMCU V1.0

nmctseb, 2017-11-24:
Can confirm 1.3.6 works without a hitch here. Win10, Arduino IDE 1.8.4 and NodeMCU v2. Using D8 for SDA/CS and D3 for RST.

heziegl, 2018-01-13:
I'm not sure if this is the same problem, but maybe this helps:

I tried to get a MFRC522 board to work with a ESP8266 Wemos D1 mini and this library version 1.3.6 (not master but release 1.3.6):

> `#define RST_PIN D3                // RST-PIN for RC522 - RFID - SPI - Modul D3/GPIO0`
> `#define SS_PIN D8                 // SDA-PIN for RC522 - RFID - SPI - Modul D8/GPIO15`
> `MFRC522 mfrc522(SS_PIN, RST_PIN); // Create MFRC522 instance`
> 
> `void setup() {`
> ...
> `SPI.begin();`
> `mfrc522.PCD_Init();`
> ...
> `}`

The call of `mfrc522.PCD_Init()` never returned.

I had to add three lines (from `MFRC522::PCD_Init()`), to make it work:
> `void setup() {`
> ...
> `SPI.begin();`
> _`pinMode(RST_PIN, OUTPUT);`_
> _`digitalWrite(RST_PIN, HIGH); // Exit power down mode. This triggers a hard reset.`_
> _`delay(50);`_
> `mfrc522.PCD_Init();`
> ...
> `}`

It seemed that otherwise the RST/D3 pin was LOW all the time and the `MFRC522::PCD_Init()` didn't reset the MFRC522.

Rotzbua, 2018-01-14:
Should be fixed with feda7b55f984c0030ab0eac2d573161815ce3ceb .

closed 2018-01-14 by Rotzbua