miguelbalboa/rfid #327: How to get accesscontrol example work properly?

henryprobosantoso, 2017-07-21:
hi, i'm henry
I am new here and i am new in programming

### Step 2: Describe your environment

 * OS version: window 7 ultimate
 * Arduino IDE version: 1.6.10
 * MFRC522 Library version: latest
 * Arduino device: uno 
 * MFRC522 device: _____
  
### Step 3: Describe the problem
i try example AccessControl from library RFID. The code I have uploaded and run, I tried it but then stopped as shown in the picture
[image]

Then I close the monitor serial and scan the master card, but I get stuck again when I try to scan other cards besides the master card.
[image]

  
#### Steps to reproduce:

  1. I tried serial monitor many times
  
#### Expected Results:

I want the program to run properly

Rotzbua, 2017-07-22:
The example was created by @omersiar and updated by @beyondszine . Maybe they can help. Otherwise you have to find it out yourself. Best.

henryprobosantoso, 2017-07-22:
Huhuhu, I have tried many ways, but I do not dare to change the code, this code is too complicated for me, considering I'm still very beginner.
Thanks before..

omersiar, 2017-07-22:
Fresh board's eeprom may need to be erased. either use wipe button or eeprom_clean example from IDE. This should solve your problem.

henryprobosantoso, 2017-07-22:
I have tried wipe button many times and not work, It just makes me repeat choose mastercard.

But i will try eeprom cleann, thank youu senpaiii .....
( :

omersiar, 2017-07-22:
No, you need to press wipe button *before* powering on arduino. and keep it
pressed

henryprobosantoso, 2017-07-22:
Mr @omersiar , thank youu , i try eeprom clear and it work, im so happy.
I dont understand about press wipe before powering arduino, it means no power ? if there is no power how can it work? and i dont understand why my eeprom full, But I just this time using the example of access control program

omersiar, 2017-07-22:
nice, your are asking good questions, keep up learning.

step by step you will learn all the details. try examples, and look for more at the forum also check out arduino reference pages also playground.

henryprobosantoso, 2017-07-22:
thank youuuuu.....

henryprobosantoso, 2017-07-22:
@omersiar , heiii , excuse me , 
Can i make my rfid tag data to be the default or delete all data that i have input to be the original

omersiar, 2017-07-22:
i did not understand what you mean, however you can always wipe all the
data, or you can re-program your master card.

henryprobosantoso, 2017-07-22:
Noo , i try the other example from library, the example make me can write my name to the rfid card when it scanned, but i dont know how to erase the data that i wrote

omersiar, 2017-07-22:
okay,  i believe there is no erase so you need to re-write  "0" other than your name, GitHub issues maintained for the Library issues itself. You can ask your questions on other platforms. Everytime you post a question we got an email about it. we can lose track here. You can close this issue now, thank you.

Rotzbua, 2017-07-25:
@henryprobosantoso If you have a better description text for the example, please make a pull request. Thanks.

closed 2017-07-25 by Rotzbua