miguelbalboa/rfid #328: USART interrupt issue

Freebornrus, 2017-07-23:
Hi! I`m new to GitHub, so I hope this is the right place to describe the problem with the library.

Environment:
 * OS version: Windows 8.1
 * Arduino IDE version:1.8.3
 * MFRC522 Library version: v1.3.6
 * Arduino device: Arduino Uno
 * MFRC522 device: RFID-RC522
  
The problem: it is impossible to use USART RX interrupt with MFRC522 library, because (like Arduino IDE say) it is already in use in this library. Try code below and you will get an error.
I`ve searched throught whole library but found nothing about using USART interrupt.
So the question is: it is really impossible to use USART_RX_vect with this library or there is any workaround?

```arduino
#include <MFRC522.h>

void setup() {}

void loop() {}

ISR(USART_RX_vect) {}
```

Rotzbua, 2017-07-23:
Correct the mfrc-library do not use `USART_RX_vect`. Maybe it is used by some Arduino libraries. If you search https://github.com/arduino/Arduino/search?utf8=%E2%9C%93&q=USART_RX_vect&type= , then you get a found in hardwarserial. So I suggest, that is a conflict with `Serial.print()`. Best.

Freebornrus, 2017-07-23:
@Rotzbua 
I wish that could be so simple, but it is not. 
Remove "#include <MFRC522.h>" from the code above and the error will not appear.
More to say: I`ve successful use of USART interrupts in other Arduino projects where I do not use MFRC522.h so that is definetly not problem with hardwareserial.

Rotzbua, 2017-07-23:
@Freebornrus You are wrong.

This will fail:
```arduino
void setup() {
  Serial.begin(9600);
}

void loop() {}

ISR(USART_RX_vect) {}
```
Result:
`(.text+0x0): multiple definition of `__vector_18'`

And yes, this library has `Serial.print()` for debugging purpose in the main code (bad but kept for legacy).

Freebornrus, 2017-07-23:
@Rotzbua 
[image]
[image]

I don`t use any Serial in code (pic #1), so there should not be problem.
Yes, I have found `Serial.print()` in library too, but all of them, as I see, commented. And no Serial.begin. 
Any included libraries can use Serial? I already tried to remove all Arduino.h, but no result.

Rotzbua, 2017-07-23:
If any `Serial` is used Arduino ide automatically include `HardwarSerial0.h` which use `USART_RX_vect`. The mfrc522 library has `Serial` in the code. So you can not use interrupt and library at the same time. Or remove very `Serial` by your self.

closed 2017-07-23 by Rotzbua

Freebornrus, 2017-07-23:
@Rotzbua 
And it worked for me!
I much appreciate your help, thank you a lot.