miguelbalboa/rfid #329: Sombody knows why this happens with my RFID card?

Perhapsody, 2017-07-25:
I was just playing arround with the RC522 trying to copy a Bus Card that we use here in Argentina.
When i run the InfoDump this pops up no matter the distance i place the card.  **sib**
[image]

Does this mean that the blocks are empty? or just i cant read them for a reason i cant understand?

Sorry for my noobness and the bother, im just playing around xD. 
(Initially my idea was to clone the Bus card, thus having a backup of it. But i think i need a UID Changeable card to make this possible)

Rotzbua, 2017-07-25:
The sectors are protected with a key. You can not copy them unless you have this keys. There is no way to get this keys with this library.

closed 2017-07-25 by Rotzbua

Carlokko, 2017-08-22:
I was doing just the same thing with my Italian Bus Card and got the same error.  @Rotzbua, @Perhapsody, any ideas on what library I could use to get the keys?

Rotzbua, 2017-08-23:
Restoring the keys to get access to protected data would be illegal. Issue locked.