miguelbalboa/rfid #335: c++11 compiler error

rogerrnichols, 2017-08-15:
### Step 1: Are you in the right place?
GitHub issues only for **bugs** or **improvements** of the library. All other topics, e.g. hardware/programming, will be closed.

For general support from the community, see [Arduino Forum](https://forum.arduino.cc/) or [StackOverflow](https://stackoverflow.com/questions/tagged/mifare).

### Step 2: Describe your environment

 * OS version: _____
 * Arduino IDE version: _____
 * MFRC522 Library version: _____
 * Arduino device: _____
 * MFRC522 device: _____
  
### Step 3: Describe the problem

#### Affected file(s) or example(s):
  
  * Filename 1
  * Filename 2
  
#### Steps to reproduce:

  1. _____
  2. _____
  3. _____
  
#### Observed Results:

  * What happened?  This could be a description, log output, etc.
  
#### Expected Results:

  * What did you expect to happen?
  
#### Relevant Code:

```arduino
  // TODO(you): code here to reproduce the problem
```

rogerrnichols, 2017-08-15:
I am using trying to compile the DumpInfo example for an Arduino Uno.

Rotzbua, 2017-08-15:
@rogerrnichols please edit your first post and fill in the required informations. thanks.

rogerrnichols, 2017-08-16:
Thanks for any help.

Windows 10
RFID version 1.3.6
Arduino 1.8.3
Uno
MFRC 522 (generic,, but not attached for sketch compile test)

I tried copying the unzipped master to the libraries and then including a
zipped file.
Both attempts to Verify/compile failed with the c++11 error message.

Rotzbua, 2017-08-17:
I could not reproduce your error. Please add a screen shot of the error. Thanks.

rogerrnichols, 2017-08-17:
Arduino: 1.8.3 (Windows 10), Board: "Arduino Uno"

WARNING: Category '' in library EEPROM is not valid. Setting to
'Uncategorized'
WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library SoftwareSerial is not valid. Setting to
'Uncategorized'
WARNING: Category '' in library Wire is not valid. Setting to
'Uncategorized'
Warning: platform.txt from core 'Arduino AVR Boards' contains deprecated
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags}
{compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}",
automatically converted to
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags}
{compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider
upgrading this core.
In file included from
C:\Users\Roger\Documents\Arduino\libraries\rfid-master\src/MFRC522.h:78:0,

                 from
C:\Users\Roger\Documents\Arduino\libraries\rfid-master\examples\DumpInfo\DumpInfo.ino:37:

C:\Users\Roger\Documents\Arduino\libraries\rfid-master\src/require_cpp11.h:9:2:
error: #error "This library needs at least a C++11 compliant compiler,
maybe compiler argument for C++11 support is missing or if you use Arduino
IDE upgrade to version >=1.6.6"

 #error "This library needs at least a C++11 compliant compiler, maybe
compiler argument for C++11 support is missing or if you use Arduino IDE
upgrade to version >=1.6.6"

  ^

exit status 1
Error compiling for board Arduino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Rotzbua, 2017-08-17:
Strange, maybe you have an old toolchain installed. "tools" -> "board manager" -> look for update

`rfid-master` seems you also did not used the library manager. your description says `RFID version 1.3.6` which is wrong. just use releases https://github.com/miguelbalboa/rfid/releases

closed 2017-08-21 by Rotzbua