miguelbalboa/rfid #339: Change return value on PCD_DumpVersionToSerial if comm fails

CarloMara, 2017-09-14:
Hi Folks,

While testing the library on an ESP32, I found out  _PCD_DumpVersionToSerial_ which is very nice function, because it allows us to see weather or not the comms are working correctly.

I'd like to add a different return value if the test fails, so that the program flow could be easily changed.

If it's ok for everyone I could make a pull request or send a patch file here.

Cheers,
Carlo

Rotzbua, 2017-09-14:
Hi,
`PCD_DumpVersionToSerial` is just for debugging by serial output. So no return should be provided.
Best

Rotzbua, 2017-09-19:
@CarloMara I thought about your suggestion. I think you want `void PCD_Init();` to return the status of the initialization process?
So a change to `bool PCD_Init();` where return of `true` indicates a successful initialization and false an error?
Otherwise you may use `bool PCD_PerformSelfTest();` to check the mfrc522.
Best regards

CarloMara, 2017-09-25:
Yes, you stole my words. 

From what I gather by looking at some examples `PCD_PerformSelfTest();` returns the same thing if the chip doesn't respond or it's an unknown firmware.

I would have imagined that `bool PCD_Init();` could give us some finer controls, like telling me that we can communicate, even if it can't detect the version of the firmware.

Thanks,
Carlo

Rotzbua, 2017-09-28:
I guess maybe the code from `MFRC522::PCD_DumpVersionToSerial` is enough:
```c++
byte v = PCD_ReadRegister(VersionReg);
// When 0x00 or 0xFF is returned, communication probably failed
if ((v == 0x00) || (v == 0xFF))
return false;
return true;
```

But maybe there is a better way? Would be good if you can check the mfrc522 documentation. Your pull request will be welcome.

szotsaki, 2018-04-10:
I checked the documentation and I'm not sure I found a better way.

Those that I found:
- `Status2Reg::ModemState[2:0]`: the problem with this it can be in any state, if a PICC is nearby, not just in `idle`.
- `ErrorReg`: mostly for PICC related errors.
- Test buses would be an overkill and for those we have to access to pin `AUX1` or `AUX2`.

Rotzbua, 2020-07-25:
Implemented this for `PCD_Init` at the fork.

closed 2020-07-25 by Rotzbua