miguelbalboa/rfid #341: RC522S RFID card

ingfranco, 2017-10-12:
Hi
Please, someone can tell me if we can use RC522S readers (not MFRC522) with Arduino Controller?
I need a reader with separate antenna, and on Aliexpress / Ebay this reader is the only one that have separate antenna. I try a sample sketch with arduino DUE but if I connect MFRC522 reader, all is OK, but if I change reader with RC522S,  I cannot read the same tag / fob.
Thank's for replay
Franco

Here is the reader I will use
[image]

Rotzbua, 2017-10-12:
Reported firmware version? Test with `firmware_check.ino`?

ingfranco, 2017-10-13:
Hi, Rotzbau

The firmware_check.ino report is:

Found chip MFRC522 Firmware ver. 0xFF.
Digital self test by MFRC522 failed.

Rotzbua, 2017-10-13:
Ok, then the arduino can not communicate with the module. Firmware ver. 0xFF or 0x00 means that something went wrong.

Rotzbua, 2017-12-03:
@ingfranco Did you found a solution?

closed 2018-01-14 by Rotzbua