miguelbalboa/rfid #347: Not working on SAMD21 based boards (MKR1000, Feather M0, Zero)

PTS93, 2017-11-22:
* OS version: Ubuntu 17.04
 * Arduino IDE version: 1.8.3
 * MFRC522 Library version: master
 * Arduino device: Adafruit Feather M0
 * MFRC522 device: RFID-RC522
  
I'm trying to use the library with the Feather M0 which uses the SAMD21 (ARM Cortex M0+) also present on all newer Arduinos like the MKR1000 and Zero.
Unfortunately the library fails to communicate with the RC522. I tested the same board with an Arduino Nano which worked fine. Any ideas why it doesn't work on the Feather?
Anything SPI wise that is not using standard SPI.h function calls that causes the incompatibility?

Rotzbua, 2017-11-22:
Can you try `1.6.3` https://github.com/miguelbalboa/rfid/releases ?

PTS93, 2017-11-22:
I tried 1.3.6 and 1.3.5, same issue.

PTS93, 2017-11-22:
Just to complete the documentation this is my wiring:
https://imgur.com/a/AKhrP

I checked it 5 times now, seems all in order.

I'm trying out the DumpInfo example.

Rotzbua, 2017-11-22:
- [x] Ok, wiring looks good.
- [x] According to your report it not depends on recent changes.
- [ ] Check out changes by #344 
- [ ] Maybe the pin config cause your error noticed in https://github.com/miguelbalboa/rfid/pull/344#issuecomment-340320910

---
General note for migrating to feather m0: https://learn.adafruit.com/adafruit-feather-m0-basic-proto/adapting-sketches-to-m0

PTS93, 2017-11-22:
Doing a hard reset as suggested in that bug report indeed did the trick for me. Maybe that is something that should be added to PCD_Init() by default? Or is there another reason why there is deliberately no hard reset? If so maybe an optional "do hard reset" argument for PCD_Init() might be the way to go?

Rotzbua, 2018-01-14:
Should be fixed with feda7b55f984c0030ab0eac2d573161815ce3ceb .

closed 2018-01-14 by Rotzbua

blooem, 2018-02-05:
I am having the same problem.

Using latest version of code
Arduno 1.8.5
Mac OSX

Just using standard ReadWrite Example works fine on a 32u4 board, however on the M0 it does not display much.
Here is wiring (Although it works on 32u4 so I know its not that): https://photos.app.goo.gl/WJMeFrJUu7ADLxKQ2
Here is a snippet from arduino serial output
[image]

Any help would be appreciated!
-Brandon

Rotzbua, 2018-02-05:
> Using latest version of code

@blooem Version of the code: git master or 1.3.6?

blooem, 2018-02-05:
@Rotzbua 
[image]

blooem, 2018-02-08:
I am willing to pay for support if anyone is interested. This is the only library that I have even remotely got to work, and really need to finish this project.

Thanks,
B

PTS93, 2018-02-08:
You should try the latest version from git not 1.3.6. The thing described here is already fixed.