miguelbalboa/rfid #348: dump RSSI?

lewisuan, 2017-11-28:
### Step 1: Describe your environment

 * OS version: macOS 10.13.1
 * Arduino IDE version: 1.8.5
 * MFRC522 Library version: mast-branch?
 * Arduino device: uno r3
 * MFRC522 device: funduino
  
### Step 2: Describe the problem

is it possible to get RSSI information from RC522? Must I use a cottonwood like this: http://www.instructables.com/id/Arduino-to-Cottonwood-UHF-RFID-Reader-RSSI-Command/?

Rotzbua, 2017-11-29:
Just look into the [data sheet](https://www.nxp.com/docs/en/data-sheet/MFRC522.pdf), I did not found anything about a rssi-information.
So no, it is not possible.

closed 2017-11-29 by Rotzbua

lewisuan, 2017-11-29:
There are some information about rssi, i just do not know how to get it. Page 59, section 9.3.3.6.

Theoretically, the chip has to identify the signal strength and then read the data. Section 9.3.3.6 talks about signal strength threshold, and it is possible to do some setting in arduino library to change the threshold.

I just want to find a way to obtain the detected signal strength. Maybe change the RC522 circuit?

Rotzbua, 2017-11-29:
Page 59, section 9.3.3.6: means how much power is used to send the signal.
The section 9.3.2.9 RxThresholdReg register offers some information about the signal strength. But it is just the `minimum` strength and it seems that the measured strength is not written to any register.

lewisuan, 2017-11-29:
Is it possible to start the minimum strength with a small value and increase it with a interval? Once we could read the data inside ID card, once stop the increasing. This may give us the minimum strength required to read the data and this will also be very helpful for my research.

PS: I am a physics+chemistry guy. I'm not familiar with the chip or coding.

lewisuan, 2017-11-29:
Problem solved with this: https://github.com/miguelbalboa/rfid/issues/43

Use PCD_WriteRegister function and loops could done my idea to get RSSI, approximately.