miguelbalboa/rfid #349: How to change pin configuration for Arduino Mega

skumpy24, 2017-12-02:
### Step 1: Describe your environment

 * OS version: Windows 10
 * Arduino IDE version: 1.8.5.
 * MFRC522 Library version: master-branch 
 * Arduino device: Arduino Mega Device
 * MFRC522 device: Chinese clone
  
### Step 2: Describe the problem

Hey guys,

at the moment there is a standard configuration given for the arduino mega. 

So it is as following:

RST - Pin 5
SS - Pin 53
MOSI - Pin 51
MISO - Pin 50
SCK - Pin 52

How can I change the Pin that i want to use? This should not be too difficult to change, but I really don't know where exactly. 

P.S.: The system is running properly, only want to change pin configuration!!

Kind regards,

skumpy24

Rotzbua, 2017-12-02:
The readme says:

> (1, 2) Configurable, typically defined as RST_PIN in sketch/program.
> (1, 2) Configurable, typically defined as SS_PIN in sketch/program.

All other pins can not be changed.

skumpy24, 2017-12-02:
Danke für die schnelle Antwort;)

Liebe Grüsse!

closed 2017-12-02 by Rotzbua