miguelbalboa/rfid #355: [Bug] No timeout with PCD_Reset

Rotzbua, 2018-01-14:
### Step 1: Describe your environment

 * MFRC522 Library version:  1.3.6 

### Step 2: Describe the problem

#### Affected file(s) or example(s):
  
  * MFRC522.cpp
  
#### Steps to reproduce:

  1. use software reset
  2. hardware fault -> no correct value
  3. infinite loop
  
#### Expected Results:

  * a timeout and optional exception
  
#### Relevant Code:

https://github.com/miguelbalboa/rfid/blob/a23f5511bcb25d6fb24831de1fde5ccd80891948/src/MFRC522.cpp#L237

#### Solution

Add timeout and change function to:
```c++
MFRC522::StatusCode MFRC522::PCD_Reset()

psgreco, 2018-01-30:
@Rotzbua this is what I did to my code to fix this issue (patch attached as txt, can't do a PR right now)
[pcd_reset.txt](https://github.com/miguelbalboa/rfid/files/1678099/pcd_reset.txt)

Rotzbua, 2018-01-31:
@psgreco Thanks, I will have a look.

commit 5879252f (2018-02-25): fix #355 possible infinite loop PCD_Reset()

commited by @psgreco
closed 2018-02-25 by Rotzbua