miguelbalboa/rfid #359: RFID on Heltec WiFi LoRa 32

IvanDreamer, 2018-02-08:
**improvements**

 * OS version: MAC OS
 * Arduino IDE version: 1.8.6
 * MFRC522 Library version: 1.3.6 
 * Arduino device: ESP32 Heltec WiFi LoRa 32 
 * MFRC522 device
  
I'd like to make work Heltec WiFi LoRa 32  board and RFID together, but did not find any working advice how to make it. 

Could someone help with it?

Rotzbua, 2018-02-08:
This module use a normal esp. You have to take the latest version from git, because 1.3.6 has problems with the esp, see #324 .
The rest should work straight forward.

IvanDreamer, 2018-02-09:
Thank you @Rotzbua! At least I got 

```
⸮Firmw⸮⸮⸮Version: 0x12 = (unknown)
Scan PICC to see UID, SAK, type, and data blocks...
```
But atm cannot read RFID tag. 

I'm trying with following code: 

```
#include <SPI.h>
#include <MFRC522.h>

//constexpr uint8_t RST_PIN = 21;          // Configurable, see typical pin layout above
//constexpr uint8_t SS_PIN = 5;         // Configurable, see typical pin layout above
#define RFID_SDA 5
#define RFID_SCK 18
#define RFID_MOSI 23
#define RFID_MISO 19
#define RFID_RST 22

//#define RFID_SDA 21
//#define RFID_SCK 5
//#define RFID_MOSI 27
//#define RFID_MISO 19
//#define RFID_RST 16

MFRC522 mfrc522(RFID_SDA, RFID_RST);  // Create MFRC522 instance

void setup() {
	Serial.begin(9600);		// Initialize serial communications with the PC
	while (!Serial);		// Do nothing if no serial port is opened (added for Arduinos based on ATMEGA32U4)
	SPI.begin(RFID_SCK, RFID_MISO, RFID_MOSI);
	mfrc522.PCD_Init();		// Init MFRC522
	mfrc522.PCD_DumpVersionToSerial();	// Show details of PCD - MFRC522 Card Reader details
	Serial.println(F("Scan PICC to see UID, SAK, type, and data blocks..."));
}

void loop() {
	// Look for new cards
	if ( ! mfrc522.PICC_IsNewCardPresent()) {
    //Serial.println(F("Looking ... "));
    //delay(1000);
		return;
	}

	// Select one of the cards
	if ( ! mfrc522.PICC_ReadCardSerial()) {
		return;
	}

	// Dump debug info about the card; PICC_HaltA() is automatically called
	mfrc522.PICC_DumpToSerial(&(mfrc522.uid));
}
```

paulmassen, 2018-02-21:
@IvanDreamer 
Had the same problem and I spent a long time looking for solutions and the solutions was in this thread!
I just downloaded the zip from master, and used your code with your pins, and it worked!
Hope you got it working too, but thanks!

IvanDreamer, 2018-02-22:
@paulmassen Glad to hear that it worked for you! 
So you got a proper RFID dump?

paulmassen, 2018-02-22:
yes, I got a proper rfid dump just like on my esp8266. As I really used your code as is, you sure it is not because of a problem with your reader?

Now, next step for me is trying to use lora to send data of rfid.

IvanDreamer, 2018-02-22:
@paulmassen hmm 
My reader works on Arduino UNO, but does not dump a tag info on ESP32. Maybe you are right and I just need to check/change hardware. 

P.S. 
Ping me when you will get working code LORA + RFID. When I was trying to do it I noticed that MISO and MOSI for LORA have different pins. SO I could not make LORA and RFID together.

paulmassen, 2018-02-22:
yes, I saw that too, but I managed to change pins 19 (which is used by lora) to pin 15 for MISO and rfid works, but not lora.

I'm a beginner with arduino, and I read somewhere that we should use I2c instead, but I'm not sure about how to do that.

I will keep you posted

paulmassen, 2018-02-22:
Ok, so I think the problem is related to SPI. 
It looks like it is not possible to have two SPIs running at the same time, but Rfid needs to initialize with SPI.begin and so does Lora.

IvanDreamer, 2018-02-22:
I'm also beginner with Arduino.

Basically, as I understood, if to make it work from the same SPI, it's possible to switch devices by SS (SDA) pin.

http://www.learningaboutelectronics.com/Articles/Multiple-SPI-devices-to-an-arduino-microcontroller.php

paulmassen, 2018-02-22:
Ah, good find. To not spam this repo with unrelated discussion, I thought we might continue discuss this here: https://github.com/paulmassen/heltec-wifi-lora/issues/1
and share our finds.

Maybe this will attract some others users too.

Rotzbua, 2018-02-22:
@IvanDreamer Correct. This library manages the ss pin, but not every library does this. So you have to check what the lora lib does.

IvanDreamer, 2018-02-22:
@Rotzbua good point, I need to check it. 

@paulmassen yes, I think we need to move our discussion to your thread, but still I did not find anything new so far.

paulmassen, 2018-02-22:
I'm currently trying to use SPI.begin on RFID, and when a new card is detected, it triggers SPI.begin on lora pins instead, when the message is sent, it goes back to SPI.begin with Rfid reader.

I feel like it might work. 
@Rotzbua any insight maybe? Thanks!

paulmassen, 2018-03-19:
Hi @IvanDreamer did you find something new on this?

IvanDreamer, 2018-03-20:
Hi @paulmassen! Unfortunately ATM don't have time.

infu, 2018-03-26:
@paulmassen @IvanDreamer thanks guys for figuring out most of it. Here is the code that makes RFID and LORA work on Heltec and TTGO WIFI LORA ESP32

`#include <SPI.h>
#include <LoRa.h>
#include <Wire.h>  

#define LORA_SCK     5    
#define LORA_MISO    19   
#define LORA_MOSI    27 
#define LORA_SS      18  
#define LORA_RST     14   
#define LORA_DI0     26  
#define LORA_BAND    868E6

#include <SPI.h>
#include <MFRC522.h>

#define RFID_SDA 5 
#define RFID_SCK 18 
#define RFID_MOSI 23
#define RFID_MISO 19
#define RFID_RST 22

MFRC522 mfrc522(RFID_SDA, RFID_RST);  // Create MFRC522 instance

int current_spi = -1; // -1 - NOT STARTED   0 - RFID   1 - LORA

void setup() {
  Serial.begin(9600);   // Initialize serial communications with the PC
  while (!Serial);    // Do nothing if no serial port is opened (added for Arduinos based on ATMEGA32U4)
  
}

void loop() {
  bool card_present = RFID_check();
  if (card_present) LORA_send();
}

void spi_select(int which) {
     if (which == current_spi) return;
     SPI.end();
     
     switch(which) {
        case 0:
          SPI.begin(RFID_SCK, RFID_MISO, RFID_MOSI);
          mfrc522.PCD_Init();   
        break;
        case 1:
          SPI.begin(LORA_SCK,LORA_MISO,LORA_MOSI,LORA_SS);
          LoRa.setPins(LORA_SS,LORA_RST,LORA_DI0);
        break;
     }

     current_spi = which;
}

int RFID_check() {
  spi_select(0);
  // Look for new cards
  if ( ! mfrc522.PICC_IsNewCardPresent()) {
    delay(100);
    return false;
  }

  // Select one of the cards
  if ( ! mfrc522.PICC_ReadCardSerial()) {
    return false;
  }

  // Dump debug info about the card; PICC_HaltA() is automatically called
  mfrc522.PICC_DumpToSerial(&(mfrc522.uid));
  
  return true;
}

void LORA_send() {
  spi_select(1);
  
  Serial.println("LoRa Sender Test"); 
  
  if (!LoRa.begin(868E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }

  Serial.println("init ok");
 
  delay(1500);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");

  LoRa.endPacket();

  Serial.println("sent one");
  delay(2000);                       // wait for a second
 }
  
`

Your pins worked out. I am not sure how you selected them? Before that I tried using the MOSI, MISO, CLK from the schematic (also used by LORA) and I didn't get the RFID to run at all.

There is probably a better way to do it and have both modules work at the same time, without having to switch with SPI.begin and SPI.end the way I am doing it. But for that I guess we need to have same MOSI, MISO and CLK. But well, I guess this one is good enough. Currently it sends LORA msg when RFID chip is nearby. I guess the only thing thats left is to try and see if we can receive LORA messages while scanning for RFID chips

paulmassen, 2018-03-27:
excellent @infu can't wait to try this! I will let you know how it works!

IvanDreamer, 2018-04-04:
Now I can confirm: It WORKS!
@infu Amazing work! You will be remembered 🥇

1nformatica, 2018-04-08:
Many thanks to @infu working for me too!

x8973, 2018-06-29:
My problem was solved as follows. I connected the MFRC522 to the same GPIO conclusions to which the LoRa chip is connected, but changed only the output number of SlaveSelect. By default, the LoRa SlaveSelect output was in a low state, which is why LoRa interfered with the work of the MISO line. I installed her SlaveSelect output to a high level, and the SPI earned, as it should.

adrianoms2004, 2019-01-22:
Ola senhores, segue codigo ajustado para teste do mfrc522 e heltec lora 

/*
 * --------------------------------------------------------------------------------------------------------------------
 * Example sketch/program showing how to read data from a PICC to serial.
 * --------------------------------------------------------------------------------------------------------------------
 * This is a MFRC522 library example; for further details and other examples see: https://github.com/miguelbalboa/rfid
 * 
 * Example sketch/program showing how to read data from a PICC (that is: a RFID Tag or Card) using a MFRC522 based RFID
 * Reader on the Arduino SPI interface.
 * 
 * When the Arduino and the MFRC522 module are connected (see the pin layout below), load this sketch into Arduino IDE
 * then verify/compile and upload it. To see the output: use Tools, Serial Monitor of the IDE (hit Ctrl+Shft+M). When
 * you present a PICC (that is: a RFID Tag or Card) at reading distance of the MFRC522 Reader/PCD, the serial output
 * will show the ID/UID, type and any data blocks it can read. Note: you may see "Timeout in communication" messages
 * when removing the PICC from reading distance too early.
 * 
 * If your reader supports it, this sketch/program will read all the PICCs presented (that is: multiple tag reading).
 * So if you stack two or more PICCs on top of each other and present them to the reader, it will first output all
 * details of the first and then the next PICC. Note that this may take some time as all data blocks are dumped, so
 * keep the PICCs at reading distance until complete.
 * 
 * @license Released into the public domain.
 * 
 * Typical pin layout used:
 * -----------------------------------------------------------------------------------------
 *             MFRC522      Arduino       Arduino   Arduino    Arduino          Arduino
 *             Reader/PCD   Uno/101       Mega      Nano v3    Leonardo/Micro   Pro Micro
 * Signal      Pin          Pin           Pin       Pin        Pin              Pin
 * -----------------------------------------------------------------------------------------
 * RST/Reset   RST          9             5         D9         RESET/ICSP-5     RST
 * SPI SS      SDA(SS)      10            53        D10        10               10
 * SPI MOSI    MOSI         11 / ICSP-4   51        D11        ICSP-4           16
 * SPI MISO    MISO         12 / ICSP-1   50        D12        ICSP-1           14
 * SPI SCK     SCK          13 / ICSP-3   52        D13        ICSP-3           15
 */

#include <SPI.h>
#include <LoRa.h>
#include <Wire.h>  

#define LORA_SCK     5    
#define LORA_MISO    19   
#define LORA_MOSI    27 
#define LORA_SS      18  
#define LORA_RST     14   
#define LORA_DI0     26  
#define LORA_BAND    868E6
#define PABOOST true  //Sem conhecimento dessa variavel mas ela deve aparecer para funcionar

#include <SPI.h>
#include <MFRC522.h>

#define RFID_SDA 5 
#define RFID_SCK 18 
#define RFID_MOSI 23
#define RFID_MISO 19
#define RFID_RST 22

MFRC522 mfrc522(RFID_SDA, RFID_RST);  // Create MFRC522 instance

int current_spi = -1; // -1 - NOT STARTED   0 - RFID   1 - LORA

void setup() {
	Serial.begin(9600);		// Initialize serial communications with the PC
	while (!Serial);		// Do nothing if no serial port is opened (added for Arduinos based on ATMEGA32U4)
	SPI.begin(RFID_SCK, RFID_MISO, RFID_MOSI);			// Init SPI bus
	mfrc522.PCD_Init();		// Init MFRC522
	mfrc522.PCD_DumpVersionToSerial();	// Show details of PCD - MFRC522 Card Reader details
	Serial.println(F("Scan PICC to see UID, SAK, type, and data blocks..."));
}

void loop() {
	// Look for new cards
	if ( ! mfrc522.PICC_IsNewCardPresent()) {
		return;
	}

	// Select one of the cards
	if ( ! mfrc522.PICC_ReadCardSerial()) {
		return;
	}

	// Dump debug info about the card; PICC_HaltA() is automatically called
	mfrc522.PICC_DumpToSerial(&(mfrc522.uid));
}

closed 2020-07-25 by Rotzbua

bhavancee, 2021-02-01:
> My problem was solved as follows. I connected the MFRC522 to the same GPIO conclusions to which the LoRa chip is connected, but changed only the output number of SlaveSelect. By default, the LoRa [...]

Hello
can you explain how you did it

Victorio1209, 2022-02-10:
Hello! How I can write the receiver? I have this code, but not received anything.

`#include <SPI.h>
#include <LoRa.h>
  

#define LORA_SCK     5    
#define LORA_MISO    19   
#define LORA_MOSI    27 
#define LORA_SS      18  
#define LORA_RST     14   
#define LORA_DI0     26  
#define LORA_BAND    915E6

void setup() {
  Serial.begin(9600);   // Initialize serial communications with the PC
  while (!Serial);    // Do nothing if no serial port is opened (added for Arduinos based on ATMEGA32U4)
  SPI.begin(LORA_SCK,LORA_MISO,LORA_MOSI,LORA_SS);
  LoRa.setPins(LORA_SS,LORA_RST,LORA_DI0);
}

void loop() {
  
  if (!LoRa.begin(915E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }

  Serial.println("init ok");
 
  delay(1500);
 int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");

    // read packet
    while (LoRa.available()) {
      Serial.print((char)LoRa.read());
    }

    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());
  }
   }`