miguelbalboa/rfid #360: Not working on STM32F205 M3

glowingsteam, 2018-02-14:
### Step 1: Describe your environment

 * OS version: Windows 10 64-bit
 * Arduino IDE version: 1.8.5
 * MFRC522 Library version: v1.3.6 from Github releases 
 * Arduino device: Adafruit WICED Feather - https://www.adafruit.com/product/3056
 * MFRC522 device: Sunfounder RFID-RC522
  
### Step 2: Describe the problem

I am trying to load the readAndWrite example yet the error I get is as follows:
[image]

I presume there is probably nothing to be done about this, and I searched around the web with no luck. If there's any fixed please let me know. : )
-B

Rotzbua, 2018-02-14:
You have to enable c++**11** support. It's a compiler flag. I think that helps you to google it or ask adafruit how to enable c++11.

Rotzbua, 2018-02-14:
Or take an older release of this library without c++11, have a look at the change log.

closed 2018-02-14 by Rotzbua