miguelbalboa/rfid #361: How to change the default key in the blocks?

AbrahamIE, 2018-02-19:
### Step 1: Describe your environment

 * OS version: W10
 * Arduino IDE version: 1.8.5
 * MFRC522 Library version: 1.3.5 
 * Arduino device: Arduino UNO
 * MFRC522 device: RC522
  
### Step 2: Describe the problem
I've tried to change the default keys of 2 tags. The first one using the read/write example by modifying the block 63 (now I can't access any block) and then with another TAG I made the same but with block 3 and now I cant access sector 0
#### Affected file(s) or example(s):
  
**I copied the example to change the code modifying this
(before overwrite block 63)**
   key.keyByte[0] = 0xFF;
   key.keyByte[1] = 0xFF;
   key.keyByte[2] = 0xFF;
   key.keyByte[3] = 0xFF;
   key.keyByte[4] = 0xFF;
   key.keyByte[5] = 0xFF;
//........
    byte sector         = 0; //
    byte blockAddr      = 3; //
    byte dataBlock[]    = {
        0x00, 0x00, 0x00, 0x00, //  I copied the first 10 WORDS as they were at the beginning
        0x00, 0x00, 0xFF, 0x07, //   
        0x80, 0x69, 0x01, 0x02, //  Then where I saw the key (FFFFFFFFFFFF, last 6 words) 
        0x03, 0x04, 0x05, 0x06  // I suposed I set my new KEY (010203040506)
    };
    byte trailerBlock   = 3; //
//..................
**After Overwritig the block 3 I changed this (suposing I was going to be able to enter with my new key)**

   key.keyByte[0] = 0x01;
   key.keyByte[1] = 0x02;
   key.keyByte[2] = 0x03;
   key.keyByte[3] = 0x04;
   key.keyByte[4] = 0x05;
   key.keyByte[5] = 0x06;
//.....................
    byte sector         = 0; //
    byte blockAddr      = 1; //
    byte dataBlock[]    = {
        0x00, 0x00, 0x00, 0x00, //  Using diferent values
        0x00, 0x00, 0xFF, 0x07, //   
        0x80, 0x69, 0x01, 0x02, //  
        0x03, 0x04, 0x05, 0x06  //
    };
    byte trailerBlock   = 1; //
  
#### Steps to reproduce:
shown upthere  
#### Observed Results:
Tag where I modifyed Block 63 and trying to read/write again with the modified code
[image]
And this one when I modified block 3 (in the other TAG)
[image]

  
#### Expected Results:

I suposed that if I modified the block replacing the first 10 words with the same value they had before and changing just the last 6 with my new password, setting up the new bytes at the beggining I was going to be able to read them with out problems
  
#### Relevant Code:
Before messing my tag
```arduino
  /**
 * ----------------------------------------------------------------------------
 * This is a MFRC522 library example; see https://github.com/miguelbalboa/rfid
 * for further details and other examples.
[... 411 lines cut ...]
        Serial.println(F("  perhaps the write didn't work properly..."));
    }
    Serial.println();
        
    // Dump the sector data
    Serial.println(F("Current data in sector:"));
    mfrc522.PICC_DumpMifareClassicSectorToSerial(&(mfrc522.uid), &key, sector);
    Serial.println();

    // Halt PICC
    mfrc522.PICC_HaltA();
    // Stop encryption on PCD
    mfrc522.PCD_StopCrypto1();
}

/**
 * Helper routine to dump a byte array as hex values to Serial.
 */
void dump_byte_array(byte *buffer, byte bufferSize) {
    for (byte i = 0; i < bufferSize; i++) {
        Serial.print(buffer[i] < 0x10 ? " 0" : " ");
        Serial.print(buffer[i], HEX);
    }
}
```

Rotzbua, 2018-02-20:
You set a new key, ok.
You have to post the relevant code how you have overwritten the old key. Nobody will read thousands line of code...
I guess you messed something up with the access bits.

closed 2018-02-24 by Rotzbua

jonmorad, 2018-06-04:
how to correct keyB  and access key ?

dorindabassey, 2019-07-17:
hello, am also having same problem,  I have changed the authentication key successful but I am not able to read data from the blocks I changed keys

28nivak, 2021-08-02:
I have one problem, i have changed the default factory key and put my own key, after that i can see the sector 3 and whole block of data but i cannot see the sector 3 data in bottom its show only PCD_Authenticate() failed: Timeout in communication.

please give any solution.

Scan a MIFARE Classic PICC to demonstrate read and write.
Using key (for A and B): 00 00 00 00 00 00
BEWARE: Data will be written to the PICC, in sector #0
Card UID: 86 0C F8 29
PICC type: MIFARE 1KB
Authenticating using key A...
Reading data from block 15 ...
Data in block 15:
 00 00 00 00 00 00 FF 07 80 69 FF FF FF FF FF 3F

Current data in sector:
   3     15   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF 3F  [ 0 0 1 ] 
         14   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         13   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         12   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 

Card UID: 86 0C F8 29
Card SAK: 08
PICC type: MIFARE 1KB
Sector Block   0  1  2  3   4  5  6  7   8  9 10 11  12 13 14 15  AccessBits
  15     63   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         62   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         61   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         60   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
  14     59   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         58   30 2C 31 2C  32 2C 33 2C  34 2C 35 2C  36 2C 37 2C  [ 0 0 0 ] 
         57   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         56   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
  13     55   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         54   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         53   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         52   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
  12     51   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         50   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         49   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         48   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
  11     47   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         46   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         45   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         44   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
  10     43   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         42   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         41   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         40   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
   9     39   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         38   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         37   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         36   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
   8     35   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         34   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         33   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         32   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
   7     31   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         30   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         29   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         28   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
   6     27   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         26   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         25   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         24   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
   5     23   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         22   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         21   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         20   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
   4     19   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         18   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         17   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         16   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
   3     15  PCD_Authenticate() failed: Timeout in communication.
   2     11  PCD_Authenticate() failed: Timeout in communication.
   1      7  PCD_Authenticate() failed: Timeout in communication.
   0      3  PCD_Authenticate() failed: Timeout in communication.