miguelbalboa/rfid #365: New license

Rotzbua, 2018-03-27:
I now maintain this library for tree years and invested a lot of time in this library. I'm not really happy with the current unlicense. Unlicense is not a open source license. That's the reason why I would like a change to a real open source license. At the moment it would be GNU Lesser 2.1 like all other standard arduino libraries.
The change would be with version 2.0.

Are there any other opinions?

TheNitek, 2018-03-27:
Why aren't you happy? Did any problems occur? I really appreciate people putting their work under unlicense - Not because I don't to give them any credit or anything, but you just don't have to worry about it when using it in your project. If you are doing bigger projects and try to honor all the libs licenses it can get quite hard and annoying.
Wouldn't something less restrictive (MIT?) be still ok?
Why LGPL2.1 and not the most current V3?

From a legal standpoint I even doubt you can change the licensing without asking everyone who contributed to this project (which is of course stupid ...)

Rotzbua, 2018-10-04:
Thanks for your feedback.

>  Why LGPL2.1 and not the most current V3?

Most of the default libraries use it. They use the extended version ` and later` , so it would also be valid to use it under the V3 conditions.

> From a legal standpoint I even doubt you can change the licensing without asking everyone who contributed to this project (which is of course stupid ...)

Yes and no. Yes, they never signed a CLA Commit License Agreement but No because they committed code under the Unlicense which make no restriction to any licensing.

Never mind: I will not change the license of this repository. 

I have done a lot of refactoring which would brake backwards compatibility. So I will only publish my fork under a new license.

closed 2018-10-04 by Rotzbua