miguelbalboa/rfid #368: MFRC522 read anomalies

mh512, 2018-04-03:
I'm wondering if there's anyone that may be able to shed some light on a issue I'm having with an Arduino Mega and MFRC522 NFC reader.

Before I describe the particular issue I'll point out that the code works 99.9% of the time with no issues and I'm hoping somone can help me refine it to prevent the issue I have occasionally.

Within my sketch I have a byte array named:

`uint8_t TagRead[4];                                          
`There are also a number of arrays storing the card UIDs:

```
uint8_t RED1[4] = { 0x17, 0xF6, 0xE2, 0xA4 };
uint8_t RED2[4] = { 0x77, 0xF9, 0xD5, 0xA4 };
uint8_t RED3[4] = { 0x85, 0x83, 0xB1, 0xFB };
```
Within the loop section of my code I do this:

```
void loop() {
    do {
      readflag = checkread();              //Returns 1 if the MFRC522 
                                             detects a card
    }
    while (!readflag);

    NewTone(readerbuzzer, 1000, 180);      //Beep
    recordid();                            //Record the scanned card's ID
    action();                              //Do something
}
```
checkread() looks like this:

```
int checkread() {
  if ( ! mfrc522.PICC_IsNewCardPresent()) {
    return 0;
  }
  mfrc522.PICC_HaltA();
  return 1;
}
```
So basically, if a card is detected it returns 1.

recordid() looks like this:

```
void recordid() {
  mfrc522.PICC_IsNewCardPresent();
  mfrc522.PICC_ReadCardSerial();
  for (int i = 0; i < mfrc522.uid.size; i++) {
    TagRead[i] = mfrc522.uid.uidByte[i];
  }
  mfrc522.PICC_HaltA();
}
```
So I end up with the scanned card's UID in the RagRead array. With this I can perform some action, for example compare it to one of the cards that's declared already:

```
void action() {
  if (*((uint32_t *)TagRead) == *((uint32_t *)RED1)) {
      dosomething();    
  }
  else if (*((uint32_t *)TagRead) == *((uint32_t *)RED1)) {
      dosomethingelse();    
  }
}
```
This all works OK 99% of the time.

The problem

Sometimes, if a card is scanned but removed too quickly, the system detects a new card (and so triggers recordid() and action()...) but the new card's UID is not recorded in TagRead[]. TagRead[] keeps the value of the previous card and so the action for the card scanned previously is triggered again.

Does anyone know why this is happening and more importantly, how I can prevent it?

If the card isn't scanned properly, could I call another function that warns the user they need to scan again?

The most important thing to solve is the calling of the action function if TagRead[] hasn't been updated with the new UID.

Is it possible that the card is detected (enough to break out of the loop) but the card info is not being obtained?

Thank you in advance.

Rotzbua, 2018-04-04:
@mh512 Complete environment information is missing. Please update your issue according to the template https://github.com/miguelbalboa/rfid/blob/master/.github/ISSUE_TEMPLATE.md . Thanks.

closed 2018-04-09 by Rotzbua

mauferrari, 2018-04-10:
Hi @mh512 , although you didn't give us more information about your system enviroment, as suggested by @Rotzbua , I have a suggestion to you.

The time wasted by calling `NewTone()` function probably is your issue.

The processor spends 180ms to produce the beep, giving the user the feedback that the card was already validate, but it was not... the system didn't read the card ID yeat.

So, I suggest you change the order, to read the card ID as fast as possible before you sound the beep!

In such a manner, the system get the new ID before the user think to take off the card from the reader. The 180ms look like few time, but as the user start listening the beep, he will start taking off the card from the reader and most people will not wait to finish the 180ms beep to do that.

Try this change on your `loop()` section:
```c
void loop() {
  do {
    readflag = checkread();    //Returns 1 if the MFRC522  detects a card
  }
  while (!readflag);
  
  recordid();                            //Read the card ID first
  NewTone(readerbuzzer, 1000, 180);      //after sound the Beep
    action();                             //Do something
}
```
I also sugest a change in function `recordid()` to became a litlle faster to read the card ID:
```c
void recordid() {
  // mfrc522.PICC_IsNewCardPresent();    // maybe, don't need to re-check the card presence here
  mfrc522.PICC_ReadCardSerial();
  for (int i = 0; i < mfrc522.uid.size; i++) {
    TagRead[i] = mfrc522.uid.uidByte[i];
  }
  mfrc522.PICC_HaltA();
}
``` 

I hope it could help you!