miguelbalboa/rfid #369: How to change RFID Card Security A key  &  B key

webinteger, 2018-04-03:
I have change keys  at chip  from the factory (FFFFFFFFFFFFh),
from arduino uno + RC522,  give pls an example, thanks!

Rotzbua, 2018-04-04:
There is a example, if i remember correctly...

closed 2018-04-04 by Rotzbua

webinteger, 2018-04-04:
Where is example, show link, example

Rotzbua, 2018-04-04:
Folder "examples"

Krupson, 2018-04-24:
In which example sketch? I don't see such example that demonstrates writing keys in "examples" folder.

umursengul, 2018-04-24:
From Arduino IDE;
File > Examples > Examples from Custom Libraries > MFRC522 > ...
You should find an example for this situation if I understand correctly.

Krupson, 2018-04-24:
Yeah, but which one from MFRC522 examples? I have checked all of them and there is nothing about changing key A and key B - that's the point.

iZhangHui, 2019-02-02:
@webinteger @Krupson Could you help to test my  PR #454  for change the default keys?