miguelbalboa/rfid #370: Use PN532

supersjimmie, 2018-04-09:
### Step 1: Describe your environment

 * OS version: Win10
 * Arduino IDE version: 1.6.11
 * MFRC522 Library version: Master (from feb 25)
 * Arduino device: ESP8266
 * MFRC522 device: PN532
  
### Step 2: Describe the problem

Firmware Version: 0x0 = (unknown)
WARNING: Communication failure, is the MFRC522 properly connected?

#### Affected file(s) or example(s):
  
Dumpinfo.ino
  
#### Steps to reproduce:

Connected the PN532 to the ESP8266 according to the ESP8266's SPI Hardware defined pins
RST to D4 and SS to D3.
Uploaded the dumpinfo.ino

When I upload a sketch with the regular libraries for the PN532 from adafruit, I am able to scan a card.
```
#include <SPI.h>
#include "PN532_SPI.h"
#include "PN532.h"
#include "NfcAdapter.h"

PN532_SPI interface(SPI, D3);
NfcAdapter nfc = NfcAdapter(interface); // create an NFC adapter object
```
The only difference is that I use the PN532 libs, and those libs do not use the RST (which I tried on D4 and D0).
  
#### Observed Results:
```
Message: Firmware Version: 0x0 = (unknown)
WARNING: Communication failure, is the MFRC522 properly connected?
```
  
#### Expected Results:

The card that I try to scan.
  
#### Relevant Code:

```arduino
/*
 * Typical pin layout used:
 * -----------------------------------------------------------------------------------------
 *             MFRC522      Arduino       Arduino   Arduino    Arduino          Arduino     ESP8266
 *             Reader/PCD   Uno/101       Mega      Nano v3    Leonardo/Micro   Pro Micro   NodeMCU
 * Signal      Pin          Pin           Pin       Pin        Pin              Pin         Pin
 * --------------------------------------------------------------------------------------------------
 * RST/Reset   RST          9             5         D9         RESET/ICSP-5     RST         var/D4
 * SPI SS      SDA(SS)      10            53        D10        10               10          var/D3
 * SPI MOSI    MOSI         11 / ICSP-4   51        D11        ICSP-4           16          GPIO13/D7
 * SPI MISO    MISO         12 / ICSP-1   50        D12        ICSP-1           14          GPIO12/D6
 * SPI SCK     SCK          13 / ICSP-3   52        D13        ICSP-3           15          GPIO14/D5
*/

#include <SPI.h>
#include <MFRC522.h>

constexpr uint8_t RST_PIN = D4;          // Configurable, see typical pin layout above
constexpr uint8_t SS_PIN = D3;         // Configurable, see typical pin layout above

MFRC522 mfrc522(SS_PIN, RST_PIN);  // Create MFRC522 instance
```

Rotzbua, 2018-04-10:
This library do not work with the PN532. You have to use another library.

closed 2018-04-10 by Rotzbua

supersjimmie, 2018-04-10:
Sorry, I was confused by this text in the readme:
```
Need more?

    If software: code it and make a pull request.
    If hardware: buy a more expensive like PN532 (supports NFC and many more, but costs about $15).

```
That suggests that a PN532 would also be possible.

Rotzbua, 2018-04-10:
Thanks. I extended the readme.