miguelbalboa/rfid #373: Is it possible to configure all other pins with ESP8266?

nardev, 2018-04-11:
Is it possible to make different wiring for all pins or just RST and SS?

Is this schema possible with NODEMCU/ESP8266???

SDA --> GPIO04
SCK --> GPIO05
MOSI --> GPIO08
MISO --> GPIO06
IRQ --> NC
GND --> GND
RST --> GPIO0
3.3V --> 3.3V

???

Rotzbua, 2018-04-11:
For general support from the community, see [Arduino Forum](https://forum.arduino.cc/) or [StackOverflow](https://stackoverflow.com/questions/tagged/mifare).

Or in your case to a esp forum.

closed 2018-04-11 by Rotzbua

nardev, 2018-04-11:
It's more about this library, is it possible to instantiate it with different GPIOs?

Rotzbua, 2018-04-11:
In general no.

At master-branch I tried something. There it is possible to set a pointer to a spiclass, but I never tested it and no example exists.
```c++
MFRC522(const byte chipSelectPin, const byte resetPowerDownPin,
SPIClass *spiClass = &SPI, const SPISettings spiSettings = SPISettings(SPI_CLOCK_DIV4, MSBFIRST, SPI_MODE0))
```