miguelbalboa/rfid #374: mfrc522 not working, but works in a Raspberry pi 3

xetum, 2018-04-14:
Hi, I'm trying to execute Dumpinfo and I get:

Firmware Version: 0x0 = (unknown)
WARNING: Communication failure, is the MFRC522 properly connected?
Scan PICC to see UID, SAK, type, and data blocks...

connections as per miguelbalboa/rfid on an arduino uno. L-shaped pin strip soldered and directly connected (without voltage divider) with dupont cables to arduino.

mfrc522.PCD_PerformSelfTest() always retuns false

The thing is that same rfid board reads 4-byte uids perfectly when connected to a Raspberry-pi 3 according to atitan/MFRC522 in ruby.

Pointers? Thanks

Rotzbua, 2018-04-15:
@xetum  Complete environment information is missing. Please update your issue according to the template https://github.com/miguelbalboa/rfid/blob/master/.github/ISSUE_TEMPLATE.md . Thanks.

shreyas-sachan, 2018-04-19:
Environment:
       OS version: ubuntu 16.04
       Arduino IDE version: 1.8.5
       MFRC522 Library version: 1.4.0
       Arduino device: arduino uno R3 / Arduino Mega 2560 Rev3
       MFRC522 device: 
Problem:
       When i try to execute Dumpinfo i get:

       Firmware Version: 0x92 = v2.0
       Scan PICC to see UID, SAK, type, and data blocks...
       Firmware Version: 0x92 = v2.0
       Scan PICC to see UID, SAK, type, and data blocks...
       Firmware Version: 0x92 = v2.0
       Scan PICC to see UID, SAK, type, and data blocks...

It keeps printing this message(without printing the card details) when I put a tag over it. Although it used to work perfectly 2-3 days ago. 

I have already checked the connections multiple times.

Rotzbua, 2018-04-20:
>    Firmware Version: 0x92 = v2.0
>   Scan PICC to see UID, SAK, type, and data blocks...

That's good, now a connection to the reader works.
Example `firmware_check` also should work.

But why the reader do not recognize the card I do not know. The only remaining thing is from troubleshooting:

> Increase the antenna gain per firmware: mfrc522.PCD_SetAntennaGain(mfrc522.RxGain_max);

xetum, 2018-04-22:
I've tried 1.4.0 both directy and with the help of a level shifter on SDA,SCK,MOSI and RST lines, no changes

    Firmware Version: 0x0 = (unknown)
    WARNING: Communication failure, is the MFRC522 properly connected?
    Scan PICC to see UID, SAK, type, and data blocks...

shreyas-sachan did you connect directly or through a level-shifter?

The interesting point is that I can read uids with a RPi3. Perhaps my arduino uno R3 is faulty?

shreyas-sachan, 2018-04-23:
@xetum I connected it directly. Although it worked perfectly fine 4-5 times. But after that, it started misbehaving.

Rotzbua, 2018-04-25:
@xetum Thanks for your feedback :)

closed 2018-04-25 by Rotzbua