miguelbalboa/rfid #377: Multiple Readers - Potential REAL fix!!! Need help finishing it!

dgriggs, 2018-04-24:
Okay, so years ago I figured out how to use multiplexers to trick the Arduino into working in rapid sequence with as many readers as you want, depending on how fast you needed to read them. But now I'm working on a project that requires a full library-level fix, not just a workaround.

I tried using this solution and it worked! Unlike with Balboa's library where a random number of readers work, and the others never respond (seemed to look a lot like SPI data interference), I get FLAWLESS response from 6 separate reader cards every time! 
https://arduino.stackexchange.com/questions/18313/issue-sharing-miso-with-multiple-rc522-rfid-readers

I believe it has something to do with the fundamental SPI setup - with the Sunfounder library using softSPI, you can set any SPI pins you want for each library instance, meaning I could change my wiring to not use SDA (slave select) and instead assign different pins to each reader's MISO.

I NEED YOUR HELP! I'm a noob at Github and I'm still not confident in writing complex library code, so hopefully someone more skilled can take the direction I've found and pull it through to the end for the community's benefit!

Option 1: Unfortunately, the library from Sunfounder which communicates flawlessly with multiple readers does not include functions for authenticating with tags to allow read/write access to data blocks. It only is good for reporting UIDs. I have tried and tried to get this library up to snuff with what Balboa's library can do copying his library's authentication process step-by-step, with no luck so far. If someone can create a repo for Sunfounder's rfid.

Option 2: get Balboa's library to accept different MISO pins for each reader! No guarantee that will completely solve the problem, but there's a good chance that's what's making the difference here. The easiest way I think to do this is simply switch Balboa's code from using hardware SPI over to softSPI library.

Sunfounder RFID.h library with softSPI.h dependency included: https://www.sunfounder.com/learn/download/UkZJRF9raXRfVjEuMF9mb3JfQXJkdWluby56aXA=/dispa

Rotzbua, 2018-04-24:
@dgriggs Complete environment information is missing. Please update your issue according to the template https://github.com/miguelbalboa/rfid/blob/master/.github/ISSUE_TEMPLATE.md . Thanks.

dgriggs, 2018-04-24:
OS version: Windows 10
Arduino IDE version: 1.8.5
MFRC522 Library version: 1.2.0
Arduino device: Mega2560
MFRC522 device: RFID-RC522 with S50 tags

Annaane, 2018-05-03:
Hello. I don't much know about accessing data blocks on tags for multiple readers but I can assure you that you can use miguelbalboa's library for connecting multiple readers without using a multiplexer, and they're all totally functional at the same time. I've just committed my code for four readers [https://github.com/Annaane/MultiRfid](url) , I hope it can help anyhow.

ildar, 2018-05-08:
I own a Chinese clone of the chip/board and see that it doesn't work 100%
right. I'm going to explore further and share my results. Meanwhile see
here: https://github.com/ildar/libnfc/issues/1

Annaane, 2018-05-10:
I don't think the problem is in the readers. I bought mine in a small shop next to my house for like 5$. Just make sure that your pins are very well wired.

pexes, 2018-05-19:
Hello, I also faces this problem and i found library similar to Sunfounders one. It has a dumping feature but there is no SPI thing so you can not initialize many readers, however, i think that you can rewrite it to support softspi.h . It is by following link:
https://www.bitsandparts.eu/documentation/77/Programmeer_voorbeeld.ino

pexes, 2018-05-19:
Now i'm trying to rewrite it, i added "read", "dumpHex", "auth" functions, but it cannot authenticate. Please reply this message, maybe together we will make that.

helitp, 2018-06-08:
I am using a modified MFRC522 with multiple readers support (currently only 2, can be changed in MFRC522.h #define MFRC522_MAXREADERS): http://heli.xbot.es/wp-content/uploads/2018/06/rfid-1.4.1_heli.zip
Only add a new SS pin for each reader.
Compatible with old code and simple use: new functions PCD_AddReader(int Pin) and PCD_SelectReader(int Reader) 
Example in \examples\DumpInfoTwoReaders

terencebeh, 2019-04-25:
@helitp. Couldn't get it to work. I have input the MFRC522 library using the link above. Usng Microbit + Microbit Driver Expansion Board. Both RFID Readers sharing the SCK (13), MISO (14), MOSI (15) and RST(8) PINs. Using pins 12 and 2 for SS. Please help. 

When I execute I get the following:  

21:59:51.862 -> Reader 0: Firmware Version: 0x0 = (unknown)
21:59:51.862 -> WARNIN⸮Reader 0: Firmware Version: 0x0 = (unknown)
21:59:52.974 -> WARNING: Communication failure, is the MFRC522 properly connected?
21:59:53.076 -> Reader 1: Firmware Version: 0xFF = (unknown)
21:59:53.144 -> WARNING: Communication failure, is the MFRC522 properly connected?

Sketch Code: 

#include <SPI.h>
#include <MFRC522.h>

constexpr uint8_t RST_PIN = 8;     // Configurable, see typical pin layout above
constexpr uint8_t SS_1_PIN = 12;   // Configurable, take a unused pin, only HIGH/LOW required, must be diffrent to SS 2
constexpr uint8_t SS_2_PIN = 2;    // Configurable, take a unused pin, only HIGH/LOW required, must be diffrent to SS 1

constexpr uint8_t NR_OF_READERS = 2;

byte ssPins[] = {SS_1_PIN, SS_2_PIN};

MFRC522 mfrc522[NR_OF_READERS];   // Create MFRC522 instance.

/**
 * Initialize.
 */
void setup() {

  Serial.begin(9600); // Initialize serial communications with the PC
  while (!Serial);    // Do nothing if no serial port is opened (added for Arduinos based on ATMEGA32U4)

  SPI.begin();        // Init SPI bus

  for (uint8_t reader = 0; reader < NR_OF_READERS; reader++) {
    mfrc522[reader].PCD_Init(ssPins[reader], RST_PIN); // Init each MFRC522 card
    Serial.print(F("Reader "));
    Serial.print(reader);
    Serial.print(F(": "));
    mfrc522[reader].PCD_DumpVersionToSerial();
  }
}

/**
 * Main loop.
 */
void loop() {

  for (uint8_t reader = 0; reader < NR_OF_READERS; reader++) {
    // Look for new cards

    if (mfrc522[reader].PICC_IsNewCardPresent() && mfrc522[reader].PICC_ReadCardSerial()) {
      Serial.print(F("Reader "));
      Serial.print(reader);
      // Show some details of the PICC (that is: the tag/card)
      Serial.print(F(": Card UID:"));
      dump_byte_array(mfrc522[reader].uid.uidByte, mfrc522[reader].uid.size);
      Serial.println();
      Serial.print(F("PICC type: "));
      MFRC522::PICC_Type piccType = mfrc522[reader].PICC_GetType(mfrc522[reader].uid.sak);
      Serial.println(mfrc522[reader].PICC_GetTypeName(piccType));

      // Halt PICC
      mfrc522[reader].PICC_HaltA();
      // Stop encryption on PCD
      mfrc522[reader].PCD_StopCrypto1();
    } //if (mfrc522[reader].PICC_IsNewC
  } //for(uint8_t reader
}

/**
 * Helper routine to dump a byte array as hex values to Serial.
 */
void dump_byte_array(byte *buffer, byte bufferSize) {
  for (byte i = 0; i < bufferSize; i++) {
    Serial.print(buffer[i] < 0x10 ? " 0" : " ");
    Serial.print(buffer[i], HEX);
  }
}

HibaHorya, 2021-08-04:
after around million of trying and trying I discovered that the correct way to connect 4 RFID RC522 is to put them in the same line on test board except SS pins and the code as usual is ReadUidMultiReader from RFID library ... like this :
[image]

f4hbw, 2021-08-24:
Thank you so much  !!!