miguelbalboa/rfid #383: Firmware ok, not reading cards

skyguy126, 2018-05-17:
The wiring is correct I can assure you, connected based on the diagram inside the comments on every file. I have tried with and without a logic level converter on 3 different boards. One of the boards worked for a few minutes but stopped working after. The current board I have right now passes the firmware check but does not read any cards when running the dump info example. 

Version is 0x92 and I am using a pro micro with a logic level converter. I have tried without as well and no luck.

### Step 1: Describe your environment

 * OS version: _____
 * Arduino IDE version: _____
 * MFRC522 Library version: _____ 
 * Arduino device: _____
 * MFRC522 device: _____
  
### Step 2: Describe the problem

#### Affected file(s) or example(s):
  
  * Filename 1
  * Filename 2
  
#### Steps to reproduce:

  1. _____
  2. _____
  3. _____
  
#### Observed Results:

  * What happened?  This could be a description, log output, screenshot, etc.
  
#### Expected Results:

  * What did you expect to happen?
  
#### Relevant Code:

```arduino
  // TODO(you): code here to reproduce the problem
```

Rotzbua, 2018-05-27:
@skyguy126 Did you solved your problem? Try a different tag, maybe your seller sent you some with wrong frequency.

aeklo, 2018-05-27:
Experiencing exactly the same. Have tried multiple boards, separate power supply, shielding the SPI wires, clocking down SPI bus, hard reset/soft reset, extensive range of tags.
By manipulating the MISO wire (pulling off/on), I sometimes manage to receive card dump. And strangely, then it is rock solid reading multiple tags simultaneously - until I reset the MCU.
It has led me into debugging the SPI bus, but this could be rc522 settings, software or hardware.
Anyone with any clues?

ChrisLSaar, 2018-06-11:
Have the same problem.

Hardware: ESP8266 D1 mini, Sainsmart RC522, all connections soldert with short cable.

Sometimes it runs for hours, sometimes it reads only 2 times.

Firmwarecheck still ok.

No RFID Card is found.

aeklo, 2018-06-11:
What I did, which miraculously solved my problem, was to edit a few settings.
It turned out my board used non-inverting TX2-coil, effectively cancelling the TX1-coil with standard settings.
Here is what I changed:
` mfrc522.PCD_WriteRegister(MFRC522::TxControlReg, 0x03); //Non-inverting TX2
    mfrc522.PCD_WriteRegister(MFRC522::GsNReg, 0x44); //Also try FF
    mfrc522.PCD_WriteRegister(MFRC522::CWGsPReg, 0x0F); //Also try 3F
    mfrc522.PCD_WriteRegister(MFRC522::ModGsPReg, 0x0F); //Also try //3F`

x8973, 2018-06-29:
I join the question. Reading and writing registers is correct, but the PCD_communicateWithPICC() function returns STATUS_TIMEOUT. There is an interrupt from the timer RC522. The antenna tuning code above did not help.

OS version: Win10 1803
Arduino IDE version: None, Eclipse Neon + ESP32-IDF
MFRC522 Library version: Latest, ported to ESP32-IDF
Arduino device: Heltec ESP32 LoRa board
MFRC522 device: RFID-RC522

x8973, 2018-07-04:
It turned out that the module has a very weak antenna. Sometimes it is possible to get a card, but reading the information from the card returns STATUS_TIMEOUT. How can I strengthen the antenna?
Reading is produced as follows:

```
if (reader->PICC_isCardPresent())
{
	MFRC522::PICC_UID_t cardUid;
	printf("Card present\n");
	if (reader->PICC_readCardSerial(&cardUid))
	{
		// UID
		printf("Card UID:");
		for (uint8_t i = 0; i < cardUid.size; i++)
		{
			printf(" %02X", cardUid.uidByte[i]);
		}
		printf("\n");
	}
}
```

Sometimes "Card present" is displayed in the console, but PICC_readCardSerial() returns STATUS_TIMEOUT.
Antenna adjustment:

```
void MFRC522::PCD_antennaOn()
{
	PCD_setRegisterBitMask(TxControlReg, 0x03);
	PCD_writeRegister(GsNReg, 0xFF); //Also try FF
	PCD_writeRegister(CWGsPReg, 0x3F); //Also try 3F
	PCD_writeRegister(ModGsPReg, 0x3F); //Also try //3F

	PCD_writeRegister(RFCfgReg, 0x0F);   	//RxGain = 48dB
} // End PCD_antennaOn()
```

aeklo, 2018-07-04:
Remember: Too high antenna gain may also prevent successful reading.

rfranky, 2018-07-10:
Hello! Same problem here.

Sometimes it works for 1 week, sometimes only ours. 
After some success reads, the antena will  not read any card anymore.

On this state (Not reading cards) I've tried a software resset of the arduino with no luck:
`void softwareReset() {
   asm volatile ( "jmp 0");
 }`

Board: Arduino nano rev3
MFRC522 device: RFID-RC522, Firmware 0x92
MFRC522 Library version: 1.3.6
IDE: Atom with PlatformIO

x8973, 2018-07-10:
From my experience I can say that in case of such a hangup, the soft reset of the microcontroller does not help. It is necessary to reset the microcircuit MRFC522.
I still achieved a steady reading of the serial number of the cards. I use the function of reading the serial number from another library, based on code Dr.Leong (WWW.B2CQSHOP.COM). I think that I will completely switch to it: it is simpler and it works. Now I'm working on adding the serial number rewriting functions to it.

x8973, 2018-07-10:
I would like details. What does "new ESP" mean?

rfranky, 2018-07-11:
I've tryed with a software resset because it solved a similar problem:
When i put two cards on the reader and try to read a mifare block `status = (MFRC522::StatusCode) mfrc522.MIFARE_Read(blockAddress, block, &size);` sometimes i get a `status != MFRC522::STATUS_OK`. And after that, without a softwareResset(), it doesen't read any more cards.

I've 12  RFID-RC522 (4 from amazon and 8 from alliexpress) trying to read cards 2 times per second 24/7.
the 4 of amazon did not fail or hang never (2 months working 24/7). 3 from alliexpress have this fail (3 or 4 fails/hangs per week).

Today I've changed one of this "broken" RFID-RC522 for one from amazon.

Time will say if this is only a problem of a bad manufactured RFID-RC522.

Rotzbua, 2018-10-01:
Can I close this? Seems the basic problem is (as usual) bad manufactured mfrc-boards... :disappointed:

rfranky, 2018-10-01:
Yes. After 3 months still working well

closed 2018-10-01 by Rotzbua

deybarmora, 2021-03-14:
Se que el foro esta cerrado, pero podría servirle a alguien en esta misma situación, así que aquí les comento mi experiencia:
Tuve un problema similar con un modulo MFRC522 corriéndolo en un ESP32, la comunicación SPI estaba OK, con el detalle que cuando consultaba la ganancia del RFID con rfid.PCD_GetAntennaGain() me devolvía cero, esto me daba indicios que el modulo se comunica mas no se configuraba bien, entre ensayo y error agregue tres líneas de código en el setup() antes de ejecutar el rfid.PCD_Init();  y esto soluciono mi problema, parece que el problema es con el estado lógico del pin RST, mas no estoy seguro. a mi me funciono pero me gustaría saber el trasfondo del problema, si alguien lo descubre me comenta. 

void setup(){
  
  // se valida que RST_PIN este en nivel lógico bajo
  pinMode(RST_PIN, OUTPUT);
  digitalWrite(RST_PIN, LOW);
  delay(10); 
  
  // Inicia modulo lector MFRC522    
  rfid.PCD_Init();                             
  
  /*
   * 0x00 - 18 dB, mínimo                                       
   * 0x01 - 23 dB                                               
   * 0x02 - 18 dB, 010b es un duplicado para 000b    
   * 0x03 - 23 dB, 011b es un duplicado para 001b    
   * 0x04 - 33 dB, Por defecto en la librería                
   * 0x05 - 38 dB                                               
   * 0x06 - 43 dB                                                
   * 0x07 - 48 dB, máximo                                       
  */
  rfid.PCD_SetRegisterBitMask(rfid.RFCfgReg, (0x07<<4));      // Poner ganancia de RFID en 48dB, (valor máximo)  
  
  Serial.print("Listo el Modulo RFID con ganancia = ");
  Serial.println(rfid.PCD_GetAntennaGain(),BIN);
}