miguelbalboa/rfid #384: Example code to save the card data to an array (MIFARE Ultralight)

CptAwe, 2018-05-28:
Hi,

Firstly, I'm new to github, so bare with me.

While your library is extremely useful, I found myself (and after some googling, and others) needing a way to "save" the data from a card to an array for manipulating it later.

The code below works **only** for MIFARE Ultralight cards, but should be simple to adapt it to any card.

```c
byte Card[16][4];

// fills the "Card[16][4]" with 0s
void ResetInfo(){
	for (int i=0; i<=15; i++){
		for (int j=0; j<=4; j++){
			Card[i][j]=0;
		}
	}
}

// fills the "Card[16][4]" with data
void ReadInfo() {

	ResetInfo();
  
  MFRC522::StatusCode status;

  byte buffer[18]; // 16 (data) + 2 (CRC)

  for (byte page=0; page<=15; page+=4){

    byte byteCount = sizeof(buffer);
    status=NFC.MIFARE_Read(page,buffer,&byteCount);
    if (status != NFC.STATUS_OK) {
      Serial.print(F("MIFARE_Read() failed: "));
      Serial.println(NFC.GetStatusCodeName(status));
      return;
    }

    //      [page][index]
    //       0-15   0-3
    //  Card [16]   [4]
    //
    //      0-3   page 0
    //      4-7   page 1
    //      8-11  page 2
    //     12-15  page 3
    //  buffer[16+2]

    int i_=0;
    for (int i=page; i<=page+3; i++){
      for (int j=0; j<=3; j++){
        Card[i][j]=buffer[4*i_ + j];
      }
      i_++;
    }
  }
  
  //  This is to stop the card from sending the info over and over again
  NFC.PICC_HaltA();
}

// shows the "Card[16][4]" after filling it
void ShowInfo(){

  ReadInfo();

  Serial.println("--------------------------");
  for (int i=0; i<16; i++){
    for (int j=0; j<4; j++){
      Serial.print(Card[i][j],HEX);
      Serial.print(" ");
    }
    Serial.println();
  }
  Serial.println("--------------------------");
}
```

It creates the global array "Card" and then fills it with the data of the card. This was written in a hurry, so it could be easily be improved.

Rotzbua, 2018-05-28:
Hi, 
thanks for the code.
I copied your snippet to the wiki: https://github.com/miguelbalboa/rfid/wiki/Useful-code-snippets
Please complete the environment information at the wiki page. Thanks

closed 2018-05-28 by Rotzbua

CptAwe, 2018-05-28:
Done! Thank you! Happy to help!