miguelbalboa/rfid #385: [MIFARE Ultralight] Support for writing data to MIFARE Ultralight card

CptAwe, 2018-05-28:
I don't seem to be able to write anything to an Ultralight card. I tried to use the MIFARE_Ultralight_Write() function, and later tried to write a wrapper around MIFARE_Write() but nothing was eventually written.

From what I read on the Ultralight card datasheet there seems to be no authentication mechanism, so the process should be really simple as I only need to send, for example:

write command | num of page (0 to 15) | data (4 bytes at a time)
------------ | ------------- | -------------
0xA2 | 0x06 | 0x01 0x02 0x03 0x04

Can I bypass the high-level functions of the library and directly tell the MFRC522 chip to send the above code?

Is my understanding correct? Is there an example I can use as reference or can you give me some pointers? I would really appreciate it!

Rotzbua, 2018-05-29:
Your environment information is missing. Please update your issue according to the template https://github.com/miguelbalboa/rfid/blob/master/.github/ISSUE_TEMPLATE.md . Thanks.

CptAwe, 2018-05-31:
Sorry, I forgot to tell you, I edited in the info.

CptAwe, 2018-06-08:
**Update:**

**Problem 1**
I don't know if my theory is correct. I believed that the only thing that needs to be sent is:
<write_command> <page_number> <4 bytes of data>

I was looking at the 3.9 datasheet (page 17) in which the fig. 16 is cropped. On the other hand on the 3.8 datasheet (page 18) the fig. 17 shows that CRC is needed at the end of the command. I will try to calculate the CRC and add it at the end, maybe then it will accept the write command.

**Problem 2**
The already existing functions (MIFARE_Ultralight_Write() or MIFARE_Write()) do not work. I have tried to use the PCD_MIFARE_Transceive() which adds the CRC data to the end of the command but nothing changed.

I was forced to go down a level and use the PCD_CommunicateWithPICC() function.

The result as of now for all the above functions is the same. I get the error "Timeout in communication". I can't imagine what I am doing wrong.

If anyone is interested I will post the code I have written.

More importantly, If anyone wants to help I will be more than grateful.

CptAwe, 2018-06-17:
**Small update:**

I have written this:

```c++
/* Function around MIFARE_Write() and MIFARE_Ultralight_Write().
 * It uses either the compatibility write of the Ultralight card (see 8.6.8)
 * or the usual write method (see 8.6.7)
 * 
 * @return STATUS_OK on success, STATUS_??? otherwise.
*/
MFRC522::StatusCode Write(	byte page, // Pages 0 - 15 (0x00 - 0x0F)
				byte *data, // The 4 byte data to write to the PICC
				byte data_size, // The data should be at least 4 bytes
				bool compatibility=false // Use compatibility write (true) or not (false)
			){

	MFRC522::StatusCode result;

	if (data == NULL || data_size < 4){
		return NFC.STATUS_INVALID;
	}

	if (compatibility){
		// Compatibility Write
		byte data_comp[16];
		memcpy(data_comp,data,4);
		for (byte i=4; i<16; i++){
			data_comp[i]=0;
		}
		result=NFC.MIFARE_Write(page,data_comp,16);
		// Serial.print("MIFARE_Write(): ");
		// Serial.println(NFC.GetStatusCodeName(result));
		if (result != NFC.STATUS_OK) {
			return result;
		}
	}else{
		// Normal Write
		result=NFC.MIFARE_Ultralight_Write(page,data,data_size);
		// Serial.print("MIFARE_Ultralight_Write(): ");
		// Serial.println(NFC.GetStatusCodeName(result));
		if (result != NFC.STATUS_OK) {
			return result;
		}
	}
	return NFC.STATUS_OK;
}
```

I have tested it on MIFARE Ultralight cards, which is what it was intended for. I am not sure it actually works, unfortunately, because all of my PICCs (~10) respond with "NAK". I suspect that it works, though, because the functions it uses work.

Split7fire, 2018-07-23:
@CptAwe Hello, I just tested your methods and they dont work for either. In my case, MIFARE_Write gives STATUS OK, but nothing was written. Standard ultralight method gets "Timeout".

CptAwe, 2019-01-16:
**Final Update**

I found out by reading the cards from another device that the nfc module read the type of the card wrong. The card actually requires a password so that I can write on it and also has additional information, excluding the data that the nfc module reported.

Because I spent so much effort on this project this revelation was heartbreaking, that's why I have stopped updating. I know this is not a valid excuse for not updating for so long but I wasn't expecting anyone to follow through.

Please excuse me for any inconvenience I may have caused.

closed 2019-01-16 by CptAwe