miguelbalboa/rfid #387: RFID Cloner Question

AnthonyFuller, 2018-06-09:
Quick question,

The RFID Cloner example:

https://github.com/miguelbalboa/rfid/blob/master/examples/RFID-Cloner/RFID-Cloner.ino

Do I just scan the card that i want to copy, then scan the card then it writes to it?

Thanks, Anthony.

Rotzbua, 2018-06-10:
The description says:

>  * Copy the RFID card data into variables and then 
>  * scan the second empty card to copy all the date

So the answer is: yes.

closed 2018-06-10 by Rotzbua

ebbeeh, 2018-06-27:
But it's not possible to scan a card in the options?

Rotzbua, 2018-06-27:
> But it's not possible to scan a card in the options?

What do you mean with "scan a card in the options" ?

ebbeeh, 2018-06-27:
the program gives you 3 options, but non of them scans the card

Rotzbua, 2018-06-27:
The example is a community commit, so maybe the example is not perfect.

I guess:
> 1 scan
> 2 show
> 3 copy

Pull requests to improve (translation, description) the example are welcome.