miguelbalboa/rfid #392: Example 'RFID-Cloner' prompts warning due to dash in name

richelbilderbeek, 2018-06-24:
### Step 1: Describe your environment

 * OS version: Lubuntu newest version: Linux winloos 4.15.0-23-generic #25-Ubuntu SMP Wed May 23 18:02:16 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux 
 * Arduino IDE version: 2:1.0.5-dfsg2-4.1
 * MFRC522 Library version: master 
 * Arduino device: Uno
 * MFRC522 device: N/A
  
### Step 2: Describe the problem

When I successfully install this library, most of the examples are added to the `File | Examples` menu. Except for `RFID-Cloner`, of which a warning dialog appears, as it has a dash in its name. Every restart of the Arduino IDE, this warning re-appears.

[image]

#### Affected file(s) or example(s):
  
  * None
  
#### Steps to reproduce:

  1. sudo apt install arduino
  2. Download this library as a zip, this zip will be called `rfid-master.zip`
  3. Unzip that zip. A folder `rfid-master` will be created
  4. Rename that folder to `RFID`
  5. In the Arduino IDE, click `Import Library` and select the RFID folder. The first warning about the dash in the name of the `RFID-Cloner` example appears
  6. When restarting the Arduino ID, the warning appears 

[image]

  7. It can be fixed by simply renaming the folder `RFID-Cloner` to `RFID_Cloner` and its file `RFID-Cloner.ino` to `RFID_Cloner.ino`
  
I volunteer to do a Pull Request with the renamed example folder. Just assign this Issue to me :-)

#### Observed Results:

  * What happened?  This could be a description, log output, screenshot, etc.

I get a warning that `RFID-Cloner` is a name with a dash.
  
#### Expected Results:

  * What did you expect to happen?

That I can start the Arduino IDE without warnings
  
#### Relevant Code:

```arduino
  // TODO(you): code here to reproduce the problem
  // No code is needed
```

Rotzbua, 2018-06-24:
Hi,
your IDE version is 1.0.5 right?
This version has several problems (one you mentioned), so the minimum IDE version was set to 1.6.
It is mentioned in the readme under `Compatible IDE`.

> This library works with Arduino IDE 1.6, older versions are not supported and will cause compiler errors.

Best

richelbilderbeek, 2018-06-24:
I fully agree it is well documented. 

I suggest (and volunteer) to do a simple rename, so that older IDE versions don't have this warning. I feel the costs outweigh the benefits, but of course, I do have less understanding of the consequences of such a rename.

Rotzbua, 2018-06-24:
It would not compile anyway, because version 1.0.X has no enabled c++11 support.

richelbilderbeek, 2018-06-24:
Yes, you are right.

The goal of documenting it is to prevent other people wasting their time in finding this out.

closed 2018-06-25 by Rotzbua