miguelbalboa/rfid #394: Logic error in PICC_Select in MFRC522.cpp

dboyz, 2018-06-25:
Hi, there seems to be a logic error here https://github.com/miguelbalboa/rfid/blob/master/src/MFRC522.cpp#L740-L743

It is unfortunate that previous pull request https://github.com/miguelbalboa/rfid/pull/301 did not fully fix this logic error. This logic error may cause the wrong bits to be modified, for example if the value of `collisionPos` is 1, the LSB of Byte 1 (NVB) will be wrongly modified.

Table below shows the corresponding value of `count` and `index` for every possible value of `collisionPos`. Notice the odd values in red font each time the value of `count` is zero.

[image]

Based on Step 7 of Section 6.4.3.1 in ISO/IEC 14443-3 (Part 3), the value of `count` and `index` should correspond to the position of the first collision

> The PCD assigns NVB with a value that specifies the number of valid bits of UID CLn. The valid bits shall be part of the UID CLn that was received before a collision occurred added by a (0)b or (1)b, decided by the PCD. A typical implementation adds a (1)b.

Diagram below illustrates the memory content of `buffer` which is an array that is used to store the received UID together with other ANTICOLLISION data.

[image]

For example, if the value `collisionPos` is found to be 9, the corresponding value of `index` and `count` should be 3 and 0 respectively

[image]

As you can see, the code which I have pointed out above was clearly not able to determine the correct corresponding `index` and `count` value.

dboyz, 2018-06-25:
Also, it is worth mentioning that https://github.com/miguelbalboa/rfid/commit/767408134cd35aca89e4b926b213bac664fa2b88 has caused an infinite loop within [this](https://github.com/miguelbalboa/rfid/blob/master/src/MFRC522.cpp#L691-L759) while loop simply because the value of `currentLevelKnownBits` is not updated.

Rotzbua, 2018-06-25:
Ok, seems I have to revert https://github.com/miguelbalboa/rfid/commit/767408134cd35aca89e4b926b213bac664fa2b88 . I guess my code flow analysis plugin made a mistake and I did not recognized this `while` loop. This function is really big (and ugly)..

For the other bug: fixes are welcome :+1:

commit 9cee699b (2018-07-05): Merge pull request #396 from dboyz/master

Fix #394 logic error in PICC_Select
closed 2018-07-05 by Rotzbua