miguelbalboa/rfid #397: Module is not working

x8973, 2018-06-26:
I apologize, this question is a little off topic library. If he violates something, but you can help me, then delete it and let me know how I can get in touch with you.
I'm trying to work with the RC522 module using the ESP32 controller. Of course, your code does not fit, so I'm trying to adapt it. I have already enabled SPI, and it works, judging by logs of the logical analyzer. However, the module responds delusionally to any attempt to read its registers. Maybe there is some way to properly initialize the module, which I do not know about?
SPI parameters:
Frequency 1 MHz
SPI mode 0
The RESET output is switched immediately before the start of the operation.

Rotzbua, 2018-06-27:
@x8973 The issue template exists not for fun: https://github.com/miguelbalboa/rfid/blob/master/.github/ISSUE_TEMPLATE.md

x8973, 2018-06-27:
Ok.
1.1 Windows 10 1803
1.2 None, Eclipse Neon + ESP32-IDF
1.3 None
1.4 Heltec ESP32 LoRa board
1.5 RFID-RC522

2.
Init SPI:

	spi_device_interface_config_t devcfg = {};
	devcfg.clock_speed_hz = 1000000;               //Clock out at 1 MHz
	devcfg.mode = 0;                                //SPI mode 0
	devcfg.spics_io_num = PIN_NUM_CS;
	devcfg.queue_size = 7;                          //We want to be able to queue 7 transactions at a time
	devcfg.cs_ena_pretrans = 0;
	devcfg.flags = 0;
	//Initialize the SPI bus
	ret = spi_bus_initialize(HSPI_HOST, &buscfg, 1);
	if (ret != ESP_OK)
	{
		ESP_LOGE(__func__, "Unable init SPI bus: 0x%04X", ret);
	}
	//Attach the RFID to the SPI bus
	ret = spi_bus_add_device(HSPI_HOST, &devcfg, &spi);
	if (ret != ESP_OK)
	{
		ESP_LOGE(__func__, "Unable add device to SPI bus: 0x%04X", ret);
	}

Read firmware version:

	if (!gpio_get_level(PIN_NUM_RST))
	{
		printf("MFRC522 in power down mode. Reseting...\n");
		// Hard Reset RFID
		gpio_set_level(PIN_NUM_RST, 1);
		vTaskDelay(50 / portTICK_PERIOD_MS);
	}
	uint8_t reg = 0;
	reg = PCD_ReadRegister(spi, VersionReg);
	printf("Register %i value 0x%02X\n", VersionReg, reg);

Read PCD register:

	esp_err_t ret;
	uint8_t reg[2];
	reg[0] = Register | 0x80;
	reg[1] = 0;
	uint8_t val[2] = {0, 0};
	spi_transaction_t t;
	memset(&t, 0, sizeof(t));       //Zero out the transaction
	t.length = 16;
	t.tx_buffer = &reg;
	t.rx_buffer = &val;
	ret = spi_device_transmit(spi,&t);
	assert(ret==ESP_OK);
	return val[1];

Console output:

	MFRC522 in power down mode. Reseting...
	Register 110 value 0xDD

The logic analyzer shows the correct operation of the SPI bus, but the module responds incorrectly. After each restart, the module's response changes.
[image]

Rotzbua, 2018-06-27:
We already have a thread about heltec please use #359

closed 2018-06-27 by Rotzbua