miguelbalboa/rfid #399: Problem with mfrc522.PCD_Init() on Leonardo Pro Micro

fdetrias, 2018-07-01:
Hi guys,
I have tested DumpInfo from examples with Wemos D1 Mini and works fine. Firmware and card info are printed in serial monitor. 
But I can't get it work with Leonardo Pro Micro, no output in serial monitor....just blank. In Leonardo Pro Micro I use pin 9 for RST and 10 for SS just like described here https://github.com/miguelbalboa/rfid#pin-layout

In DumpInfo sketch I found out that seems like it stuck in mfrc522.PCD_Init(), because when I commented it out this line and reupload sketch, serial monitor can produced this output.

> Firmware Version: 0xFF = (unknown)
> WARNING: Communication failure, is the MFRC522 properly connected?
> Scan PICC to see UID, SAK, type, and data blocks...

What should I do from here ? Thanks for helping :)

Rotzbua, 2018-07-05:
A `Leonardo Pro Micro` do not exist. Please check this and complete your environment information.

Rotzbua, 2018-07-12:
C.losed because no reaction.

closed 2018-07-12 by Rotzbua