miguelbalboa/rfid #400: Firmware Version: 0x1C = (unknown) Scan PICC to see UID, SAK, type, and data blocks...

Hotchongas, 2018-07-03:
### Step 1: Describe your environment

 * OS version: _____
 * Arduino IDE version: _____
 * MFRC522 Library version: _____ 
 * Arduino device: _____
 * MFRC522 device: _____
  
### Step 2: Describe the problem

#### Affected file(s) or example(s):
  
  * Filename 1
  * Filename 2
  
#### Steps to reproduce:

  1. _____
  2. _____
  3. _____
  
#### Observed Results:

  * What happened?  This could be a description, log output, screenshot, etc.
  
#### Expected Results:

  * What did you expect to happen?
  
#### Relevant Code:

```c++
  // TODO(you): code here to reproduce the problem
```

Rotzbua, 2018-07-12:
>Firmware Version: 0x1C

If you already done everthing in the troubleshooting section of the readme, then I guess your hardware has an error. Contact your supplier.

Best

closed 2018-07-12 by Rotzbua

sridivya1, 2019-09-03:
Firmware Version: 0xEE = (unknown)
Scan PICC to see UID, SAK, type, and data blocks...
I am getting the above error. My version of mfrc522 is 1.4.2.

nvatuan, 2021-01-28:
> Firmware Version: 0xEE = (unknown)

I had this exact problem with reported version 0xEE. I redid the soldering and it worked fine now. For future readers, maybe check your soldering again.

novixe439, 2021-04-30:
I already resoldered it but it still doesn't work 😭

novixe439, 2021-04-30:
Does it mean that my hardware is the problem? Pls help me I really need this for a project 🥺

Rotzbua, 2021-04-30:
@novixe439 There is big help https://github.com/miguelbalboa/rfid#troubleshooting otherwise you have to contact your supplier.