miguelbalboa/rfid #401: working

awsmhemi, 2018-07-11:
anyone plz explain me the working of these lines... This from rfid reader sample program.

 for (byte i = 0; i < mfrc522.uid.size; i++) 
  {
     Serial.print(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " ");
     Serial.print(mfrc522.uid.uidByte[i], HEX);
     content.concat(String(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " "));
     content.concat(String(mfrc522.uid.uidByte[i], HEX));
  }

awsmhemi, 2018-07-11:
[image]

Rotzbua, 2018-07-12:
Hi,
this is a programming question. Please ask in the arduino forum, stackoverflow or something else you prefer.
Best

closed 2018-07-12 by Rotzbua