miguelbalboa/rfid #402: After PCD_PerformSelfTest() card can`t be readed

saloid, 2018-07-12:
### Step 1: Describe your environment

 * OS version: Win 10
 * Arduino IDE version: 1.9.0-beta
 * MFRC522 Library version: 1.4.0 (from arduino library manager) 
 * Arduino device: Nano
 * MFRC522 device: RFID-RC522 (blue board)
  
### Step 2: Describe the problem

After using `PCD_PerformSelfTest()` I should init (call `PCD_Init()`) module again, because without this it doesn`t reads cards (`PICC_IsNewCardPresent()` always returns "false").

#### Affected file(s) or example(s):
  
  * Filename 1
  * Filename 2
  
#### Relevant Code:

doesn`t works:
```c++
  rfid.PCD_Init();
  Serial.println(rfid.PCD_PerformSelfTest() ? "ok" : "ERROR!");
```

works: 
```c++
  rfid.PCD_Init();
  Serial.println(rfid.PCD_PerformSelfTest() ? "ok" : "ERROR!");
  rfid.PCD_Init();
```

HenrikSte, 2019-01-12:
Thanks for posting this. 
Spent an hour to find out why I am not reading anything...
Adding a init after self test fixed it.

fmuntean, 2020-05-13:
Same issue here: the SelfTest is leaving the board in an unusable state until init is called again.