miguelbalboa/rfid #403: Bug storing cards (AccessControl.ino)

Rotzbua, 2018-07-12:
### Step 1: Describe your environment

 * Arduino IDE version: 1.8.5
 * MFRC522 Library version: master
 * Arduino device: independed
 * MFRC522 device: independed
  
### Step 2: Describe the problem

#### Affected file(s) or example(s):
  
  * AccessControl.ino
  
#### Steps to reproduce:

**Copied from** #389 :
In the code when i = 1 and the count =1 (which means there is a card registered) the '<' makes the condition not to be true at that time. ' <=' resolves the issue.
  
#### Observed Results:

  * storing card id does not work correctly
  
#### Other relevant issues/pr

Solved a issue/ introduced current issue #346 
Fix of current issue but introduce old issue #389

#### Next steps

We need a good fix for both issues. Means correct handling of *extreme/special* values like `0x00 0xff`.

rafaelmalheiros, 2018-08-10:
What #346 really fixed? 
It is not clear. 
Could someone explain what was the bug? 
It says that fixes an infinite loop, but changing `i <= count` to `i < count` will never solve an infinite loop problem. 
What about `0x00 0xff`? When are they causing issues?

Without further information, looks like it fixed a issue to a very specific scenario to always make the last inserted record unreachble.

Rotzbua, 2018-08-13:
For #346 lets ping autor @mads256c .

The example is not from me. So my suggestion is just a suggestion ;).

Yes I think the 255 aka `0xff` is the problem.

I found a suspicious line of code:
 
https://github.com/miguelbalboa/rfid/blame/master/examples/AccessControl/AccessControl.ino#L373
```c++
//////////////////////////////////////// Read an ID from EEPROM //////////////////////////////
void readID( uint8_t number ) {
  uint8_t start = (number * 4 ) + 2;    // Figure out starting position
  for ( uint8_t i = 0; i < 4; i++ ) {     // Loop 4 times to get the 4 Bytes
    storedCard[i] = EEPROM.read(start + i);   // Assign values read from EEPROM to array
  }
}
```

For 255
> uint8_t start = (number * 4 ) + 2; 
> (255 *4) +2
> 1020 +2
> 1022

So start at 1022 and then we read 4 byte -> end at 1026, but we only have 1024 on arduino.

I do not have the time to review and fix the complete example. For me it is just an example. So if you @rafaelmalheiros can fix the code I appreciate that :+1: .

lmmeng, 2018-08-20:
Hi @Rotzbua,
uint8_t can't hold numbers > 255. 
So, the result of the (255*4) + 2 = 1022 = 0x3FE will be truncated to 0xFE = 254.

Regards,
Liviu

commit a93e0fab (2021-07-31): Remove example AccessControl.ino

solves #568
closes #403
closed 2021-07-31 by Rotzbua