miguelbalboa/rfid #411: Issue with detecting card not present with Array for UID's.

Monoth, 2018-08-05:
I am using OSX High Sierra.
Using Master branch of MFRC522 library with a rc522 reader.

I'm sorry to post this here but I also posted to stack overflow and the Arduino forum and I have found no solution.

I am using an array for some stored UIDs. I simply want to print a message *once only* when one of these is near the reader and *once only* when it is taken away.

This works fine when using boolean flags and no array; just by simply detecting if any card is present or not as follows:

`#include <SPI.h>
#include <MFRC522.h>
#define SS_PIN 10
#define RST_PIN 9
MFRC522 mfrc522(SS_PIN, RST_PIN);   // Create MFRC522 instance.

bool CurrentCardPresentStatus = false;         // current state of the button
bool lastCardPresentStatus = false;     // previous state of the button

void setup() {
  Serial.begin(115200); // Open serial monitor at 115200 baud to see ping results.  
  SPI.begin();      // Initiate  SPI bus
  mfrc522.PCD_Init();   // Initiate MFRC522
}

void loop(){

  CurrentCardPresentStatus = mfrc522.PICC_IsNewCardPresent();

  if (CurrentCardPresentStatus != lastCardPresentStatus) {
    
    if (CurrentCardPresentStatus == true) {
      Serial.println("CARD PRESENT");      
    } else {
      Serial.println("NO CARD");  
    }
  }
  lastCardPresentStatus = CurrentCardPresentStatus;
    
  mfrc522.PICC_HaltA();          // Halt PICC
  mfrc522.PCD_StopCrypto1();     // Stop encryption on PCD    
}`

However, when I use this method with the array, the code does not behave how I expect. I need to uncomment the line `//mfrc522.PICC_ReadCardSerial();` for the code to reach the array, and when I do that all three print messages print to console when a card is close to reader, and nothing when card is taken away.

Basically the `mfrc522.PICC_ReadCardSerial();` function  is causing the issues, but it seems it is necessary in order to get the four elements of the UID.

The code is here:

`#include <SPI.h>
#include <MFRC522.h>
#define SS_PIN 10
#define RST_PIN 9
MFRC522 mfrc522(SS_PIN, RST_PIN);   // Create MFRC522 instance.

char* instrumentUIDs[] = {"E6 7A 78 89", "67 10 DB 2B", "54 8F DA 2B", "84 C2 BE 1E", "A7 0B 75 F2"};

bool CurrentCardPresentStatus = false;         // current state of the button
bool lastCardPresentStatus = false;     // previous state of the button

void setup() {
  Serial.begin(115200); // Open serial monitor at 115200 baud to see ping results.  
  SPI.begin();      // Initiate  SPI bus
  mfrc522.PCD_Init();   // Initiate MFRC522
  Serial.println("Approximate your card to the reader...");
  Serial.println();
}

void loop(){

  CurrentCardPresentStatus = mfrc522.PICC_IsNewCardPresent();

  if (CurrentCardPresentStatus != lastCardPresentStatus) {
    
    if (CurrentCardPresentStatus == true) {
      Serial.println("CARD PRESENT (OUTSIDE OF ARRAY)");
      ObjectTagChecker();
    } else {
      Serial.println("NO CARD");  
    }
  }
  lastCardPresentStatus = CurrentCardPresentStatus;
    
  mfrc522.PICC_HaltA();          // Halt PICC
  mfrc522.PCD_StopCrypto1();     // Stop encryption on PCD  
} 

void ObjectTagChecker() {  
  //mfrc522.PICC_ReadCardSerial();   // NEED TO UNCOMMENT THIS FOR CODE TO REACH ARRAY BELOW
  String content= "";

  for (byte i = 0; i < mfrc522.uid.size; i++) {
     content.concat(String(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " "));
     content.concat(String(mfrc522.uid.uidByte[i], HEX));
  }

  content.toUpperCase();

  for (int i = 0; i < (sizeof(instrumentUIDs)/2); i++) {  
    if (content.substring(1) == instrumentUIDs[i]) {       
        Serial.println("CARD PRESENT (INSIDE OF ARRAY)");   // CODE WILL NOT REACH HERE UNLESS "mfrc522.PICC_ReadCardSerial();" ABOVE IS UNCOMMENTED
    } 
  }
}`

Again, very sorry to post this here but I've been at this for several days and can't figure it out. I know it's probably something very simple but I'd be ever so grateful if you could point out how to get around this?

Many thanks

Martin

Rotzbua, 2018-08-05:
https://github.com/miguelbalboa/rfid/blob/master/.github/ISSUE_TEMPLATE.md#step-1-describe-your-environment

Monoth, 2018-08-05:
Are those details now sufficient, Rotzbua? Thanks.

lmmeng, 2018-08-20:
Hi Martin,
in my opinion this behavior is the "expected" one - you need to read the UID before you can use it.
The PICC_IsNewCardPresent() function only tells you if a card is near the reader, it doesn't read the information on card. UID is first returned after issuing a read command - the PICC_ReadCardSerial() is doing it.

Regards,
Liviu

Rotzbua, 2018-10-01:
@Monoth Is this solved?

closed 2018-10-04 by Rotzbua

Monoth, 2018-10-04:
Yes, I solved it. Sorry for delay. Extremely busy. If you need me to post the code I can do. I don't have it to hand right now. It was quite simple in the end I think.

Rotzbua, 2018-10-05:
It would be nice if you can post your solution, but it is ok if you have no time :+1: