miguelbalboa/rfid #421: WDT reset on NodeMCU ESP8266

C0rn3j, 2018-09-28:
This may or may not have something to do with this issue: https://github.com/miguelbalboa/rfid/issues/293
```
Firmware Version: 0x92 = v2.0
Scan PICC to see UID, SAK, type, and data blocks...
Card UID: AB 96 56 D3
Card SAK: 08
PICC type: MIFARE 1KB
Sector Block   0  1  2  3   4  5  6  7   8  9 10 11  12 13 14 15  AccessBits
  15     63   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         62   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         61   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         60   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
  14     59   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         58   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         57   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         56   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
  13     55   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         54   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         53   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         52   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
  12     51   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         50   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         49   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         48   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
  11     47   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         46   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         45   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         44   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
  10     43   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         42   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         41   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         40   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
   9     39   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         38   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         37   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         36   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
   8     35   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         34   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         33   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         32   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
   7     31   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         30   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         29   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         28   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
   6     27   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         26   00 00 00 00  00 00 00 00 
Soft WDT reset

ctx: cont 
sp: 3ffffcb0 end: 3fffffc0 offset: 01b0

>>>stack>>>
3ffffe60:  00004937 3ffee678 3ffffe8e 4020429d  
3ffffe70:  00000000 00000001 00000000 00000001  
3ffffe80:  00000761 00000000 00000000 302029cc  
3ffffe90:  00000000 00000001 0000000c 00000006  
3ffffea0:  00000009 3fffff06 3ffffef0 40202e12  
3ffffeb0:  3fffff06 00000000 00000000 00000006  
3ffffec0:  3ffffef7 00000007 00000007 4020408b  
3ffffed0:  00000000 00000001 3ffee7a8 402047b0  
3ffffee0:  3ffffef8 00000008 00000008 4020374f  
3ffffef0:  00000000 00000000 00000000 00000000  
3fffff00:  00004937 00120100 3ffee7a8 402040a0  
3fffff10:  00000000 00000000 00000018 00000004  
3fffff20:  00000002 00000002 3ffee678 3ffee67c  
3fffff30:  3fffff80 3ffee67c 3ffee7a8 40204204  
3fffff40:  3ffee67c 00000010 3ffee7a8 3fffff80  
3fffff50:  3ffee67c 00000006 3ffee678 40203998  
3fffff60:  00000000 00000000 3ffee678 3ffee7e4  
3fffff70:  3ffee7a8 3ffee67c 3ffee678 40203b30  
[... 87 lines cut ...]
 * MFRC522 device: uhh Chinese MFRC522 reader
  
### Step 2: Describe the problem

#### Affected file(s) or example(s):
  
https://github.com/miguelbalboa/rfid/blob/master/examples/DumpInfo/DumpInfo.ino
  
#### Steps to reproduce:

  1. Use the example DumpInfo sketch, change RST and SS pins (D3, D1 in my case)
  2. Upload it, run it and watch serial
  3. Try various tags and they all seem to fail at a certain point
  
#### Observed Results:

As per attached log on top 
  
#### Expected Results:

The card to actually read fully without a WDT reset
  
#### Relevant Code:

DumpInfo example

C0rn3j, 2018-09-28:
Well that was a fun ride. Setting the speed from 9600 to 115200 made it work.

Rotzbua, 2018-09-29:
> Well that was a fun ride. Setting the speed from 9600 to 115200 made it work.

Correct. Increasing the speed is a valid solution. `9600` is the default speed for arduino samples. I never observed a WDT reset on a D1 Mini (on older SDKs).

closed 2018-09-29 by Rotzbua

Rotzbua, 2018-09-30:
> 9600 is the default speed for arduino samples.

The main target platform is still arduino. I wound change this. The WDT reset seems a bug in the ESP-Arduino framework https://github.com/esp8266/Arduino .