miguelbalboa/rfid #422: MFRC522Hack is not included in the Arduino library

C0rn3j, 2018-09-28:
```
c0rn3j@Aleksandra : ~/Arduino/libraries/MFRC522/src
[0] % ls -lah
total 164K
drwxr-xr-x 2 c0rn3j c0rn3j 4.0K Sep 28 16:37 .
drwxr-xr-x 5 c0rn3j c0rn3j 4.0K Sep 28 16:37 ..
-rw-r--r-- 1 c0rn3j c0rn3j  638 Nov 30  1979 deprecated.h
-rw-r--r-- 1 c0rn3j c0rn3j  73K Nov 30  1979 MFRC522.cpp
-rw-r--r-- 1 c0rn3j c0rn3j  35K Nov 30  1979 MFRC522Extended.cpp
-rw-r--r-- 1 c0rn3j c0rn3j 4.1K Nov 30  1979 MFRC522Extended.h
-rw-r--r-- 1 c0rn3j c0rn3j  25K Nov 30  1979 MFRC522.h
-rw-r--r-- 1 c0rn3j c0rn3j  385 Nov 30  1979 require_cpp11.h
```

This does not match what is in here - https://github.com/miguelbalboa/rfid/tree/master/src

Rotzbua, 2018-09-29:
Everything is right.

https://help.github.com/articles/about-releases/

closed 2018-09-29 by Rotzbua

C0rn3j, 2018-09-29:
Let me rephrase, I downloaded the library via the Arduino IDE and the files were NOT there.

Rotzbua, 2018-09-30:
Yes, correct. The latest release is 1.4.0 which can be found here: https://github.com/miguelbalboa/rfid/releases

Branch `master` is not a release, it is a `development` branch which means it is a unreleased state of the software.