miguelbalboa/rfid #430: Sudden 300ms+ delay on mfrc522.PICC_IsNewCardPresent()

Landfishy, 2018-10-22:
This is blowing my mind. I am running an arduinoMega2560 with a standard ethernet shield and the RFID-RC522. I've been programming this little guy for several weeks, and had my program working, then all of a sudden I am getting a 300ms+ lag spike whenever I call mfrc522.PICC_IsNewCardPresent(). I thought I was going crazy because I rolled back to the version of the program I was working with that morning and it was also showing the lag spike. Then I replaced every component except the Mega (spare coming in the mail), and the issue persisted, even across a new RFID-RC522. Here is the feedback from millis() on each routine of my program...

> runCheckRFID entered >
Time is ============================== 384555ms
Time is ========IsNewCardPresent========== 384855ms
< runCheckRFID exited <
> runDisplayUpdate entered >
Time is ============================== 384859ms
> manageSFXtriggers entered >
Time is ============================== 384866ms
> runLEDChangeTracker entered >
Time is ============================== 384896ms
< runLEDChangeTracker exited <
<  runDisplayUpdate exited <
> handleInboundENET entered >
Time is ============================== 384908ms
< handleInboundENET exited <
> updateInputs entered >
Time is ============================== 384917ms
< updateInputs exited <
Time is ============================== 384923ms
> runModeControl entered >
Time is ============================== 384928ms
< runModeControl exited <
> runCheckRFID entered >
Time is ============================== 384934ms
Time is ========IsNewCardPresent========== 385234ms
< runCheckRFID exited <

If I enter a run mode that does not call mfrc522.PICC_IsNewCardPresent(), the program accelerates back up to its expected cycletime. We didn't even know it was doing this until all of our lighting systems suddenly started shuddering with a refresh rate of about 3Hz.

And here is the use of the function, in the routine...
`int runCheckRFID()
{
  if(DEBUGMODE)Serial.println("> runCheckRFID entered >");
  Serial.print("Time is ============================== "); Serial.println(millis());  
  //Check to see if there is not a new card; if not, just bail out of the subroutine.
  if ( ! mfrc522.PICC_IsNewCardPresent()){ Serial.print("Time is ========IsNewCardPresent=========== "); Serial.println(millis()); return;}
  if ( ! mfrc522.PICC_ReadCardSerial()) {Serial.print("Time is ========ReadCardSerial=========== "); Serial.println(millis()); return;}
}
`
Does anyone have any idea what is going on? I am absolutely confounded right now! D:

Rotzbua, 2018-10-22:
I don't have supernatural abilities. Please add your environment information.
https://github.com/miguelbalboa/rfid/blob/master/.github/ISSUE_TEMPLATE.md

Landfishy, 2018-10-22:
My apologies for the missing info and accidental doublepost. 

Step 1: Describe your environment
OS version: Windows 7
Arduino IDE version: 1.8.5
MFRC522 Library version: 1.3.6, 1.4.1
Arduino device: Ardunio MEGA2560
MFRC522 device: RFID-RC522
Step 2: Describe the problem
So again, after working with the system for several weeks I am now experiencing an exactly 300ms long lag spike when calling the function mfrc522.PICC_IsNewCardPresent() where it used to only take a fraction of the time.  Normally I'd get a stream of rfid information out when a chip was placed atop the reader, but now it hangs, attempting to read about 3 times a second an fails to acknowledge the presence of a new card/rfid token.  
Steps to reproduce: Rolled back to a much earlier version of the program that did not have this issue and it is also exhibiting the same behavior. Replaced all hardware, including the reader, but not the arduino (because a spare is on the way in the mail), and the issue still persists. I added in code to output the time, milis(), to observe what portion of the program was hanging, and it occurs between the time when the mfrc522.PICC_IsNewCardPresent() routine is called to the time after it exits.
Observed Results:
Here is an actual output stream, showing some of the debugging information from my other routines, compared to the mfrc522.PICC_IsNewCardPresent() call. 

runCheckRFID entered >
Time is ============================== 384555ms
Time is ========IsNewCardPresent========== 384855ms
< runCheckRFID exited <
runDisplayUpdate entered >
Time is ============================== 384859ms
manageSFXtriggers entered >
Time is ============================== 384866ms
runLEDChangeTracker entered >
Time is ============================== 384896ms
< runLEDChangeTracker exited <
< runDisplayUpdate exited <
handleInboundENET entered >
Time is ============================== 384908ms
< handleInboundENET exited <
updateInputs entered >
Time is ============================== 384917ms
< updateInputs exited <
Time is ============================== 384923ms
runModeControl entered >
Time is ============================== 384928ms
< runModeControl exited <
runCheckRFID entered >
Time is ============================== 384934ms
Time is ========IsNewCardPresent========== 385234ms
< runCheckRFID exited <

Expected Results:
Expected mfrc522.PICC_IsNewCardPresent() to take less than 30ms. I never had to clock how quickly it was running before, but the serial output from the program was a high speed stream of information whenever an RFID was read. 

Plan: I am going to strip down my code tonight to just the bare essentials of the RFID code to prove that none of my other code is affecting this process. I do not know what I did to cause this delay to start happening, but it seems inherent in the software since all the hardware has been replaced. I'm also not too worried about my own, non-rfid code being a problem, because as soon as I comment out the mfrc522.PICC_IsNewCardPresent()  line, everything runs at its expected speeds with no hanging whatsoever. It super suspicious that its taking exactly 300ms every time, is this a timeout by chance?

Thanks again for the help!!

Landfishy, 2018-10-23:
Ok! I did what I said I'd do, and stripped the program down to only the way I am using the RFID reader, and it is still giving me the lag spikes. Here is the program, and here is my output:

`//Embedded Libraries
#include <SoftwareSerial.h>
#include <Math.h> 
#include <SPI.h>

//External library definitions
#include <MFRC522.h>

//----------------------------------------------------------------------//
//                         PIN INITIALIZATION                           //
//----------------------------------------------------------------------//

//DEFINE THE NUMBER OF PINS BASED ON THE INFO ABOVE
#define NUM_PINS 2

//----------------------------------------------------------------------//
//                         RFID CONFIGURATION                           //
//----------------------------------------------------------------------//
constexpr uint8_t RST_PIN = 8;         // Configurable, see typical pin layout above
constexpr uint8_t SS_PIN = 9;         // Configurable, see typical pin layout above
MFRC522 mfrc522(SS_PIN, RST_PIN);  // Create MFRC522 instance

String lastreadUID = "00 00 00 00"; //A container for the last read UID           
bool allowscans = 0; //allow or disallow RFID scanning

void setup() {
  // put your setup code here, to run once:
  //Initialize serial communications
  Serial.begin(115200);
  Serial.println("Serial Connection Started at 115200Hz successfully.");
  
  //Initialize RFID Reader
  SPI.begin();      // Initiate  SPI bus
  mfrc522.PCD_Init();   // Initiate MFRC522
  Serial.println("RFID interface started successfully");
}

void loop() {
  
  //Upate time
  unsigned long currenttime = millis();
  runCheckRFID();
}

int runCheckRFID()
{
  Serial.println("> runCheckRFID entered >");
  Serial.print("Time is ============================== "); Serial.println(millis());  
  //Check to see if there is not a new card; if not, just bail out of the subroutine.
  if ( ! mfrc522.PICC_IsNewCardPresent()){ Serial.print("Time is ========IsNewCardPresent=========== "); Serial.println(millis()); return;}
  if ( ! mfrc522.PICC_ReadCardSerial()) {Serial.print("Time is ========ReadCardSerial=========== "); Serial.println(millis()); return;}

  //Collect the UID from RFID
  Serial.print("A WILD RFID APPEARED! =");
  String content= "";
  byte letter;
  
  //Construct a single string from each byte of input from the RFID
  for (byte i = 0; i < mfrc522.uid.size; i++) 
  {
     //Serial.print(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " ");
     //Serial.print(mfrc522.uid.uidByte[i], HEX);
     content.concat(String(mfrc522.uid.uidByte[i], HEX));
     content.concat(String(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " "));
  }

  content.toUpperCase();
  Serial.println(content);
}`

Serial Connection Started at 115200Hz successfully.
RFID interface started successfully
> runCheckRFID entered >
Time is ============================== 55
Time is ========IsNewCardPresent=========== 353
> runCheckRFID entered >
Time is ============================== 356
Time is ========IsNewCardPresent=========== 654
> runCheckRFID entered >
Time is ============================== 658
Time is ========IsNewCardPresent=========== 956
> runCheckRFID entered >
Time is ============================== 960
Time is ========IsNewCardPresent=========== 1258
> runCheckRFID entered >
Time is ============================== 1262
Time is ========IsNewCardPresent=========== 1559
> runCheckRFID entered >
Time is ============================== 1563
Time is ========IsNewCardPresent=========== 1861
> runCheckRFID entered >
Time is ============================== 1865
Time is ========IsNewCardPresent=========== 2163
> runCheckRFID entered >
Time is ============================== 2167
Time is ========IsNewCardPresent=========== 2465
> runCheckRFID entered >
Time is ============================== 2469

Rotzbua, 2018-10-23:
You measure `IsNewCardPresent` + `Serial.println("==")`.

Landfishy, 2018-10-23:
> You measure `IsNewCardPresent` + `Serial.println("==")`.

Hi there Rotzbua,  thanks for your response. I am going to interpret your message as "The reason you are seeing lag is because you are running the IsNewCardPresent() routine immediately followed by a Serial print command. 

Normally this line reads... 
if ( ! mfrc522.PICC_IsNewCardPresent()) return;

The only reason I have added the serial print commands in this program are to prove exactly where the lag spike exists. In the real program, we noticed this lag problem because suddenly all the led indicators started blinking unusually. With a single serial message in the main loop which shows the program time, we still see the 300ms+ delay.

Unless you are saying something else?

Landfishy, 2018-10-24:
For your convenience, I've gone ahead and re-attached a copy of the program with only a single serial print command in the main loop to prove that the issue is still happening.

`//Embedded` Libraries
#include <SoftwareSerial.h>
#include <Math.h> 
#include <SPI.h>

//External library definitions
#include <MFRC522.h>

//----------------------------------------------------------------------//
//                         PIN INITIALIZATION                           //
//----------------------------------------------------------------------//

//DEFINE THE NUMBER OF PINS BASED ON THE INFO ABOVE
#define NUM_PINS 2

//----------------------------------------------------------------------//
//                         RFID CONFIGURATION                           //
//----------------------------------------------------------------------//
constexpr uint8_t RST_PIN = 8;         // Configurable, see typical pin layout above
constexpr uint8_t SS_PIN = 9;         // Configurable, see typical pin layout above
MFRC522 mfrc522(SS_PIN, RST_PIN);  // Create MFRC522 instance

String lastreadUID = "00 00 00 00"; //A container for the last read UID           
bool allowscans = 0; //allow or disallow RFID scanning

void setup() {
  // put your setup code here, to run once:
  //Initialize serial communications
  Serial.begin(115200);
  Serial.println("Serial Connection Started at 115200Hz successfully.");
  
  //Initialize RFID Reader
  SPI.begin();      // Initiate  SPI bus
  mfrc522.PCD_Init();   // Initiate MFRC522
  Serial.println("RFID interface started successfully");
}

void loop() {
  
  //Upate time
  unsigned long currenttime = millis();
  runCheckRFID();
}

int runCheckRFID()
{
  //Serial.println("> runCheckRFID entered >");
  //Serial.print("Time is ============================== "); Serial.println(millis());  
  //Check to see if there is not a new card; if not, just bail out of the subroutine.
  if ( ! mfrc522.PICC_IsNewCardPresent()) return;
  if ( ! mfrc522.PICC_ReadCardSerial()) return;

  //Collect the UID from RFID
  //Serial.print("A WILD RFID APPEARED! =");
  String content= "";
  byte letter;
  
  //Construct a single string from each byte of input from the RFID
  for (byte i = 0; i < mfrc522.uid.size; i++) 
  {
     //Serial.print(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " ");
     //Serial.print(mfrc522.uid.uidByte[i], HEX);
     content.concat(String(mfrc522.uid.uidByte[i], HEX));
     content.concat(String(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " "));
  }

  content.toUpperCase();
  Serial.println(content);
}`

Rotzbua, 2018-10-25:
Can reproduce this huge delay. Only observed max 30ms without token and 3 with token.

Sorry but seems that your hardware got some problems. The reader officially do not support 5V, some people reported problems. Maybe try a level shifter.
Best

### Test code
```c++
void loop() {
  unsigned long first = millis();
  bool result =  mfrc522.PICC_IsNewCardPresent();
  unsigned long second = millis();
  //if (30 < second - first ) // filter small delay
  Serial.println(second - first);
}
```

### Measurements
```
26
27
28
26
27
28
26
28
28
26
28
28
26
28
26
```

closed 2018-10-25 by Rotzbua