miguelbalboa/rfid #432: Example 'FixBrickedUID' may cause card can't be read

EndangeredF1sh, 2018-10-27:
### Step 1: Describe your environment

 * OS version: macOS 10.13
 * Arduino IDE version: 1.8.4
 * MFRC522 Library version: v1.3.6(the newest line in changes.txt)
 * Arduino device: uno
 * MFRC522 device: RFID-RC522-Firmware Version: 0x92 = v2.0
  
### Step 2: Describe the problem

#### Affected file(s) or example(s):
  
  * FixBrickedUID
  * MFRC522.cpp
  
#### Steps to reproduce:

  1. put a blank CUID card and run FixBrickedUID.
  2. use example 'ChangeUID' to write sector 0 for new UID.

#### Observed Results:

  *   RFID reader may not read the card and Android APP 'Mifare Classic Tool' will show that 'this card is not Mifare Classic Card ,etc.'
  
#### Expected Results:

  * CUID card can be read normally.
  
#### Relevant Code:

```c++
bool MFRC522::MIFARE_UnbrickUidSector(bool logErrors) {
	MIFARE_OpenUidBackdoor(logErrors);
	
	byte block0_buffer[] = {0x01, 0x02, 0x03, 0x04, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};
	
	// Write modified block 0 back to card
	MFRC522::StatusCode status = MIFARE_Write((byte)0, block0_buffer, (byte)16);
	if (status != STATUS_OK) {
		if (logErrors) {
			Serial.print(F("MIFARE_Write() failed: "));
			Serial.println(GetStatusCodeName(status));
		}
		return false;
	}
	return true;
}
```
### MORE DETAILS
A normal card sector 0 may be like this(I CHANGED THE '33 85 BF 11' PART FOR NOTHING LEAK):
```
   0      3   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
          2   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
          1   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
          0   33 85 BF 11  4E 08 04 00  01 6F 01 6D  45 68 F8 1D  [ 0 0 0 ] 
```

A CUID card after 'FixBrickedUID' sector 0 will like this:
```
   0      3   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
          2   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
          1   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
          0   01 02 03 04  04 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ]
```
if you use the 'ChangeUID', this sector 0 will change to it:
```
   0      3   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
          2   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
          1   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
          0   33 85 BF 11  04 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ]
```
so if you read the card, it may show nothing. It seems that the card is 'damaged' but it just some wrong data in sector 0.
I don't know how to fix the program, so I open the issue and modified the function 'MIFARE_UnbrickUidSector' to fix the bad card.
Sorry for bad English.

Rotzbua, 2018-10-27:
Ok. My first thought was that the byte 7-15 (=manufacturer data) are zero. So the program can not detect the manufacturer and it thinks it is a damaged card. 

https://www.nxp.com/docs/en/data-sheet/MF1S50YYX_V1.pdf
Figure 7
Byte 0-6 are uid
byte 7-15 are manufacturer data

Please notice that 2 types of uids exists: 4 byte and 7 byte. 7 byte are more common this days.

EndangeredF1sh, 2018-10-27:
> Ok. My first thought was that the byte 7-15 (=manufacturer data) are zero. So the program can not detect the manufacturer and it thinks it is a damaged card. [...]

Hi, Rotzbua
Thanks to your comment, it is really helpful.
After I change any byte in sector 0, card will not work.
So I think the example 'FixBrickedUID' write wrong data to sector 0 (it write 01 02 03 04  04 00 00 00  00 00 00 00  00 00 00 00 to it)
And example 'ChangeUID' (it write 4 byte UID to card) can't read card because of wrong manufacturer data(Byte 5-15 are all zero, but the byte 4 BBC code is right(04)) so it can't write correctly.
They lead to the problem together.
Thanks for your reading.
cheers.

closed 2018-10-29 by Rotzbua