miguelbalboa/rfid #437: Detect presence of card works only every second try

bgoeppner, 2018-11-06:
### Step 1: Describe your environment

 * OS version: Win 10
 * Arduino IDE version: 1.8.7
 * MFRC522 Library version: 1.4.3
 * Arduino device: Arduino Nano V3 (Old Bootloader)
 * MFRC522 device: AZDelivery RFID RC522
  
### Step 2: Describe the problem

Detecting if a card is present in loop yields alternately "present" and "not present" (see below for code sample and serial output).

For my mp3 player, I need to continously detect if a RFID-Card is present or not. 

Each card is linked to a mp3 file. 
If card is detected, the according mp3-file is played. 
If card is removed, mp3 is paused. 
If card is detected again, playing resumes (or starts new song if different card)

If card detection does not work reliable, the player does pause and resume all the time which makes listening a pain...

#### Steps to reproduce:

  1. Compile and upload the sketch below to Arduino nano
  2. Put RFID-card on reader
  3. Monitor serial output
  
#### Observed Results:

* With the sketch below, serial output is like following:
```
No card found
Card found:  92 25 6F EB
No card found
Card found:  92 25 6F EB
No card found
Card found:  92 25 6F EB
No card found
Card found:  92 25 6F EB
```
* When `PIC_HaltA()` and/or `PCD_StopCrypto1()` are uncommented, card uid is only printed when I put card on the reader. While on the reader, serial output  continuously yields "No card found". When card is removed and put on the Reader again, Serial Output yields the uid once, and then again continuously "No card found". ==> This is even worse.
  
#### Expected Results:

* Serial Output yields "Card found: <UID>" when card is put on the reader.
* Serial Output yields "No card found" when there is no card on the reader (and only then).
  
#### Relevant Code:

```c++
#include <stdint.h>
#include <SPI.h>
#include <MFRC522.h>

#define SS_PIN 10
#define RST_PIN 9

MFRC522 mfrc522(SS_PIN,RST_PIN);

void setup() {
  Serial.begin(112500);

  SPI.begin();        // Init SPI bus
  mfrc522.PCD_Init(); // Init MFRC522
}

void loop() {
  if (mfrc522.PICC_IsNewCardPresent() && mfrc522.PICC_ReadCardSerial()) {
    // uid/card found
    Serial.print(F("Card found: "));
    dump_byte_array(mfrc522.uid.uidByte, mfrc522.uid.size);

    // stop communication
    //mfrc522.PICC_HaltA();
    //mfrc522.PCD_StopCrypto1();
  } else {
    // no uid/card found
    Serial.println(F("No card found"));
  }
}

/**
   Helper routine to dump a byte array as hex values to Serial.
*/
void dump_byte_array(byte *buffer, byte bufferSize) {
  for (byte i = 0; i < bufferSize; i++) {
    Serial.print(buffer[i] < 0x10 ? " 0" : " ");
    Serial.print(buffer[i], HEX);
  }
  Serial.println();
}

```

bgoeppner, 2018-11-06:
#279 seems to be similar, however, the code in the Issue did not work in my case

Rotzbua, 2018-11-07:
Works as expected:
```c++
/**
 * Returns true if a PICC responds to PICC_CMD_REQA.
 * Only "new" cards in state IDLE are invited. Sleeping cards in state HALT are ignored.
 * 
 * @return bool
 */
bool MFRC522::PICC_IsNewCardPresent() 
```

Just test once a new uid is present and then repeat try reading the card until it fails..

closed 2018-11-12 by Rotzbua