miguelbalboa/rfid #438: Multiple Readers - Authentication Problems

Tomas1337, 2018-11-12:
### Step 1: Describe your environment

 * OS version: Windows 8
 * Arduino IDE version: 1.8.7
 * MFRC522 Library version: 1.4.3 
 * Arduino device: Arduino Uno
 * MFRC522 device: Genuine RC522 Board based on the [Collection] Bad boards #428 post picture
  
**Setup:**
Arduino Uno connected to 2 RC522 boards with everything in parallel except the SDA pin. Pin 10 and Pin 8 of the Arduino are attached to each SDA pin of the RC522. Everything is bread boarded so wire length shouldnt be much of an issue.

**Problem:**

I'm trying to Write into NFC card using multiple readers. For the love of me, I have not yet seen a post out there yet who's trying to do authentication using multiple readers. I've only seen getting UID posts with multiple readers. 

I've based my code off miguelbalboa's ReadUIDMultireader example and added authentication methods found on MiFareClassicValueBlock example.  The problem is, I can't get past the authentication A method. When Authenticating I get the following error:
```
The UID: AB 3F 3E E9 has been assgined to Reader 0
Authenticating using key A...
PCD_Authenticate() failed:Timeout in communication.
```

I've gotten the ReadUIDMultireader & the ValueBlock to work so it's not my hardware nor the keys that are the problem.

**Stuff i've tried:**

  1. First I thought that the data being passed into the PCD_Authenticate was incorrect or lost but I've double checked that the data is correct by serial printing out the key and the UID from the library itself when the Authenticate function is called.

 2. I thought that the waitIRq found in PCD_Authenitace was sort of a wait time for the reader to get a response and thought increasing this value would allow the reader more allowance(stupid I know). so I changed the value of it from 0x10 to 0x64. Result was that it passed the Authentications sequence, but I still wasn't able to write. The more I dig deep into the library, it seems like I just bypassed the authentication routine forcing the status to OK.
  

  
#### Relevant Code:

```c++
#include <SPI.h>
#include <MFRC522.h>

#define SS_1_PIN 10 // Attach Reader #1
#define SS_2_PIN 8 // Attach Reader #2
#define RST_PIN 9
#define NR_OF_READERS 2 // 

byte ssPins[] = {SS_1_PIN, SS_2_PIN};
int referTable[2] = {0,1};// Reference table
int lockerStatus[2] = {0,0}; //0 for Unoccuupied & 1 for Occupied
byte uidRecord[2][4]; // Array of UID to be Set
int cardBalance;
int lockerPrice = 20;
uint8_t reader;
byte sector         = 1;

MFRC522 mfrc522[NR_OF_READERS]; // Create MFRC522 instances
MFRC522::MIFARE_Key key;

void setup() {

  Serial.begin(9600); // Initialize serial communications with the PC
  while (!Serial);    // Do nothing if no serial port is opened (added for Arduinos based on ATMEGA32U4)

  SPI.begin();        // Init SPI bus

  // Prepare the key (used both as key A and as key B)
  // using FFFFFFFFFFFFh which is the default at chip delivery from the factory  
  for (byte i = 0; i < 6; i++) {
        key.keyByte[i] = 0xFF;
    }
 
    Serial.println(F("Scan a MIFARE Classic PICC to demonstrate Value Block mode."));
    Serial.print(F("Using key (for A and B):"));
    dump_byte_array(key.keyByte, MFRC522::MF_KEY_SIZE);
    Serial.println();
    
    Serial.println(F("BEWARE: Data will be written to the PICC, in sector #1"));
 
  for (reader = 0; reader < NR_OF_READERS; reader++) {
    mfrc522[reader].PCD_Init(ssPins[reader], RST_PIN); // Init each MFRC522 card
    Serial.print(F("Reader "));
    Serial.print(reader);
    Serial.print(F(": "));
    mfrc522[reader].PCD_DumpVersionToSerial();
  }
}

void loop() {
  
for (reader = 0; reader < NR_OF_READERS; reader++) {
    // Look for new cards

    if (mfrc522[reader].PICC_IsNewCardPresent() && mfrc522[reader].PICC_ReadCardSerial()) {
      Serial.print(F("Reader "));
      Serial.print(reader);
      // Show some details of the PICC (that is: the tag/card)
      Serial.print(F(": Card UID:"));
      dump_byte_array(mfrc522[reader].uid.uidByte, mfrc522[reader].uid.size);
      Serial.println();
      Serial.print(F("PICC type: "));
      MFRC522::PICC_Type piccType = mfrc522[reader].PICC_GetType(mfrc522[reader].uid.sak);
      Serial.println(mfrc522[reader].PICC_GetTypeName(piccType));
      

      // Halt PICC
      mfrc522[reader].PICC_HaltA();
      // Stop encryption on PCD
      mfrc522[reader].PCD_StopCrypto1();

//LOCKER SETUP FILE STARTS HERE

    // Check for compatibility

            byte valueBlockA    = 5;
            byte trailerBlock   = 7;
            MFRC522::StatusCode status;
            byte buffer[18];
            byte size = sizeof(buffer);
            int32_t value;

            // Authenticate using key A
                  Serial.println(F("Authenticating using key A..."));
                  status = mfrc522[reader].PCD_Authenticate(MFRC522::PICC_CMD_MF_AUTH_KEY_A, trailerBlock, &key, &(mfrc522[reader].uid));              
                  if (status != MFRC522::STATUS_OK) {
                  Serial.print(F("PCD_Authenticate() failed:"));
                  Serial.println(mfrc522[reader].GetStatusCodeName(status));
                  return;
                  }
                  
Serial.println("Authentication success");
   
    } //if (mfrc522[reader].PICC_IsNewC
  } //for(uint8_t reader

    // Dump the sector data
    //mfrc522[reader].PICC_DumpMifareClassicSectorToSerial(&(mfrc522[reader].uid), &key, sector);
    //Serial.println();

    // Halt PICC
    mfrc522[reader].PICC_HaltA();
    // Stop encryption on PCD
    mfrc522[reader].PCD_StopCrypto1();
}

/**
 * Helper routine to dump a byte array as hex values to Serial.
 */
void dump_byte_array(byte *buffer, byte bufferSize) {
  for (byte i = 0; i < bufferSize; i++) {
    Serial.print(buffer[i] < 0x10 ? " 0" : " ");
    Serial.print(buffer[i], HEX);
  }
}
```
I'd really appreciate a discussion to start on this so we can help each other out.
Any suggestions are welcome as I'm starting to get desperate here

Rotzbua, 2018-12-11:
Hi @Tomas1337 , did you found a solution for your question? Would you like to share your solution or can I close?

Tomas1337, 2018-12-13:
Unfortunately, I gave up and just resorted to modifying the project for one reader.
Couldn't find a solution

Rotzbua, 2018-12-13:
I'm sorry to hear that. Good luck for your current project. I will close this issue.

closed 2018-12-13 by Rotzbua

herbgiz, 2020-10-15:
I had the same problem, but due to time shortage I moved to using 1 rfid sensor, but after completion I moved to solving the problem of multiple rfid personal data.
Your problem was solved simply by removing
`// Halt PICC
      mfrc522[reader].PICC_HaltA();
      // Stop encryption on PCD
      mfrc522[reader].PCD_StopCrypto1();`

at line 64, i.e. above LOCKER SETUP (comments)
Still your code has some more issue , which hangs the controller.
I will be adding a pull request for examples to authenticate and read personal data from multiple rfid sensors.
Hope this helps.

wahajmurtaza, 2020-10-15:
I had the same problem, but due to time shortage I moved to using 1 rfid sensor, but after completion I moved to solving the problem of multiple rfid personal data.
Your problem was solved simply by removing

```
//Halt PICC
mfrc522[reader].PICC_HaltA();
// Stop encryption on PCD
mfrc522[reader].PCD_StopCrypto1();
```

at line 64, i.e. above LOCKER SETUP (comments)
Still your code has some more issue , which hangs the controller.
I will be adding a pull request for examples to authenticate and read personal data from multiple rfid sensors.
Hope this helps.

wahajmurtaza, 2020-10-15:
@Rotzbua @Tomas1337  I have added pull request #532 , which is working fine for me, might solve your problem