miguelbalboa/rfid #450: Board won't read tags, but gets version

avaldes95, 2019-01-23:
### Step 1: Describe your environment

 * OS version: _____
 * Arduino IDE version: _____
 * MFRC522 Library version: _____ 
 * Arduino device: _____
 * MFRC522 device: _____
  
### Step 2: Describe the problem

#### Affected file(s) or example(s):
  
  * Filename 1
  * Filename 2
  
#### Steps to reproduce:

  1. _____
  2. _____
  3. _____
  
#### Observed Results:

  * What happened?  This could be a description, log output, screenshot, etc.
  
#### Expected Results:

  * What did you expect to happen?
  
#### Relevant Code:

```c++
  // TODO(you): code here to reproduce the problem
```

Rotzbua, 2019-01-24:
https://github.com/miguelbalboa/rfid/blob/master/.github/ISSUE_TEMPLATE.md

avaldes95, 2019-01-24:
Windows 10
Arduino IDE 1.8.8
Release v1.4.1
Arduino Uno
RFID RC522

Problem: When I use the firmware check, the response is: 
Firmware 0x92 v2.0
Result: Defect or Unknown

https://github.com/miguelbalboa/rfid/blob/master/examples/firmware_check/firmware_check.ino

Then, I tried to run another one of the programs to try to scan a tag and it doesn't scan a tag or card that came with the package.

Rotzbua, 2019-01-24:
Then the firmware of your chip is damaged or maybe you got a fake mfrc522. You can not fix that. You can compare the printing on your chip with known bad chips, see #428.

avaldes95, 2019-01-24:
but it works with another library that is a branch of a previous release of this one

Rotzbua, 2019-01-25:
The `firmware_check.ino` checks the firmware of the chip according to the official data sheet. If the check fails, there must be something wrong with your chip.

There are some fake chips which do not implement all functions of the genuine chip. The reported version was `0x12` maybe they now report version `0x92`.

It is possible that the older version of this library works with this fake chip because it do not use all functions of the genuine chip.

jbarros35, 2019-02-12:
I need to know where to get the library which works. I don't need the "all functions for the real board" I only need to read tags!

closed 2019-02-14 by Rotzbua

Maclec, 2020-09-09:
MFRC522 Digital self test
*****************************
Firmware Version: 0x91 = v1.0
-----------------------------
Only known versions supported
-----------------------------
Performing test...
-----------------------------
Result: DEFECT or UNKNOWN
 I am using RFID MFRC522 and  facing this type of problem . please help me , what can i do for my project.