miguelbalboa/rfid #455: Write Error

williamesp2015, 2019-02-16:
I am using ESP32 with Arduino compiler and having problem writing to a generic MFRC522 
https://www.instructables.com/id/Arduino-RFID-Reader-MFRC522-Turorial/
I get this error:

Card UID: D0 EC ED 1B
PICC type: MIFARE 1KB
Authenticating using key A...
PCD_Authenticate() failed: Timeout in communication.
Card UID: D0 EC ED 1B
PICC type: MIFARE 1KB
Authenticating using key A...
PCD_Authenticate() failed: Timeout in communication.
Card UID: D0 EC ED 1B
PICC type: MIFARE 1KB
Authenticating using key A...
PCD_Authenticate() failed: Timeout in communication.
Card UID: D0 EC ED 1B
PICC type: MIFARE 1KB
Authenticating using key A...

PCD_Authenticate() failed: Timeout in communication.

williamesp2015, 2019-02-18:
I tested all examples of writing example and I got the same timeout error. I also  reduced SPI speed but no difference. How to check if RFID Card is locked or unwritable?

Rotzbua, 2019-02-19:
> I am using ESP32 with Arduino compiler and having problem writing to a generic MFRC522
https://www.instructables.com/id/Arduino-RFID-Reader-MFRC522-Turorial/

Your environment is not clear.

Please add the information according to the template:
https://github.com/miguelbalboa/rfid/blob/master/.github/ISSUE_TEMPLATE.md

closed 2019-02-25 by Rotzbua