miguelbalboa/rfid #456: Firmware Version: 0x15= (unknown)

adaiwa, 2019-03-06:
Hey,
problem with 0x12 is well-know, but what about 0x15? Has anyone heard about it?

I bought RC522 with chip NXT TXD6080, connected to nodemcu v3, all connections looks fine:
SDA -> D2
SCK -> D5
MOSI -> D7
MISO ->D6
RST -> D4
and 3.3v with GND

after then i run firmware_check example and this show me version 0x15 unknow.

Any idea?

Rotzbua, 2019-03-08:
Looks like this bad board: https://github.com/miguelbalboa/rfid/issues/428#issuecomment-429471060

> Any idea?

Contact seller / try to refund.

lmmeng, 2019-03-22:
Hi @adaiwa;
Is the board working OK with other, functional examples?
If yes, I suppose the firmware_check example needs an update, to include the new board.

Just my 2 (euro)cents.

closed 2019-06-03 by Rotzbua