miguelbalboa/rfid #458: ISO/IEC 14443-4 Cards, Timeout

knopserl, 2019-03-22:
I know, this was already discussed and the thread was closed, but before I give up on that and change the reader, I anted to confirm, that this is either solved or not solvable (because of the reader)

I have also many of this RC522 cards with firmware v92 version 2 which is the most recent version from NXP.
THe library is really great, but I have now new Security Chips (Cards) with obviously ISO 14443-4 type and cant communicate with the controller. I guess the same type of chips on my Maeastro and Credit Card as I have the same type and the same problem.
I have used the MFRC522Extended.h class
I get timeouts if I communicate with the cards using the proper (according to the docu) command structure.
I have used both: TCL_Transceive and PCD_TransceiveData.
I think it might be also a problem of the MFRC522 readers as they could have problems with
ISO 14443-4 (non A) cards. Can anybody confirm that or have a solution?
I have already odered other readers like PN532 which claims to read ISO 14443-4 cards.

Here my log:

UID: 0123456789abcdefaa55
Card UID: 01 23 45 67 89 AB CD EF AA 55
Card SAK: 20
PICC type: PICC compliant with ISO/IEC 14443-4
Dumping memory contents not implemented for that PICC type.

PCD_TransceiveData status=3
0000000000000000000000000000000000000000000000000000000000000000

Rotzbua, 2019-03-23:
please add the basic information from [template](https://github.com/miguelbalboa/rfid/blob/master/.github/ISSUE_TEMPLATE.md):
```
Arduino IDE version: _____
MFRC522 Library version: _____
Arduino device: _____
MFRC522 device: _____

knopserl, 2019-03-23:
@Rotzbua ok, here are the missing infos

**Arduino IDE** version: 1.8.7

**MFRC522 Library version:**  latestd master, (1.4.3) just downlaoded and installed on 22.3.2019 via Arduino Library Manager

**Arduino device:** Wemos D1 Mini (ESP8266)

**MFRC522 device:**  MFRC-522  (RFID-RC522) with NXP Chip, Firmware v92 V2.0 from Aliexpress
https://de.aliexpress.com/item/MFRC-522-RC-522-RC522-13-56-MHz-RFID-Modul-F-r-arduino-SPI-Schriftsteller-Reader/32909482286.html?spm=a2g0x.search0104.3.1.3cb06d5bH0XaRL&ws_ab_test=searchweb0_0%2Csearchweb201602_9_10065_10068_10547_319_317_10548_10696_10084_453_10083_454_10618_10304_10307_10820_10821_537_10302_536_10902_10059_10884_10887_321_322_10103%2Csearchweb201603_70%2CppcSwitch_0&algo_pvid=a7814690-355f-434c-8af9-bc7d8cb7f71f&algo_expid=a7814690-355f-434c-8af9-bc7d8cb7f71f-0

**Code:**
Attached the code of the sketch (currenty only the code parts for the RFID chip reading is active). 

[ESP8266_IFX_Card.zip](https://github.com/miguelbalboa/rfid/files/2999055/ESP8266_IFX_Card.zip)

knopserl, 2019-04-09:
I got the PN532 NFC Reader and with that, I can read the Security Card using the isDataExchange API and I get a response from the Security Controller (0x90 00 which is success)

bitkinbitki, 2019-05-07:
@knopserl I've same issue with mfrc522 IC. I want to try with PN532 reader like you do but I didn't find isDataExchange API. Could you help me?

knopserl, 2019-05-19:
@bitkinbitki I was switching to a PN532 library. This one seem to work for me https://github.com/elechouse/PN532.  At least I could communicate with the security processor on the chipcard. Take a look if this helps. 
I used the API: nfc.inDataExchange(selectApdu, sizeof(selectApdu), response, &responseLength) of this lib.

knopserl, 2019-09-22:
as this is still open and I did not give up using this very nice library, I spent another couple of hours but no success so far. I'm using an Infineon Security2Go card which I was told, that is compareable to standard credit or passport chips.
There is a detailed documentation available:
https://github.com/Infineon/Blockchain/blob/master/doc/BlockchainSecurity2Go_UserManual.pdf
I attach also the sketch and the respone I get from the PICC:

So I no longer get a command timeout but a wrong result. It should be length of 75 and according to chaper 4.3.1 (SELECT command) the last two bytes shall be 0x90 00 (success). The sketch with the PN532 returns length 75 and the last to bytes are 0x90 00.

I have tried to look at the PN532 code (which works with the PN532) but was not able to get a "translated" or adopted code to work with this libary (send command code with select statement) using this API: inDataExchange(selectApdu, sizeof(selectApdu), response, &responseLength) 

I get this as response from the Tag (which is wrong):
NFC Manager
Firmware Version: 0x92 = v2.0
waiting for cards
Card placed
Card UID: 01 23 45 67 89 AB CD EF AA 55    .#Eg.....U

nfcType: SECURITY, PICC compliant with ISO/IEC 14443-4
retlen: 64
 23 00 00 00 13 00 00 00 00 FF 00 00 00 00 FF 00 00 00 00 FF 0B 0E 0D 80 50 1F FB 3F 14 01 FC 3F FA 10 40 3F 10 00 00 00 00 FF 00 00 00 00 FF 00 00 00 00 FF 57 0E 0D 80 70 1F FB 3F 14 01 FC 3F    #.......................P..?...?..@?................W...p..?...?

Maybe we are not so far away and just a small thing is missing/wrong, or the RC522 does not read such cards (I believe it can communicate with this card, I hope).

[ESP32_NFC_RC522.zip](https://github.com/miguelbalboa/rfid/files/3639467/ESP32_NFC_RC522.zip)

Rotzbua, 2019-09-25:
Note: may related to #312.

xsulitoz, 2021-09-21:
Since you have already got the PN532 reader, would you please try it with a smartphones and see if the UID is changing or not? I will be very thankful for you if you did. please contact me on xsulitoz@gmail.com

Obsttube, 2023-03-19:
@knopserl
In line 118 in ESP32_NFC_RC522.ino from ESP32_NFC_RC522.zip
change "uint8_t dat[64];" to "uint8_t dat[75];" or maybe better to sth like "uint8_t dat[128];"
Let me know if it works now or how you ended up solving it.

TheoL0, 2025-04-08:
Hi ! I'm working on a similar project -> esp32 and mfrc522 in spi communication to collect ndef message from 14443-4 tag.

I got the UID and SAK but I try to send apdu commands to select application NDEF and read NDEF file but I receive "strange" datas like no-valid with no 0x9000 ,

There is a specific security decoding of hexadecimal data in iso14443 type 4 protocol ?