miguelbalboa/rfid #461: library not working on feather M0 LoRa

ydylwj23, 2019-04-23:
### Step 1: Describe your environment

 * OS version: Windows 10
 * Arduino IDE version: 1.8.9
 * MFRC522 Library version: 1.4.4 
 * Arduino device: feather M0 LoRa
 * MFRC522 device: MFRC522 chip(bought on Amazon)
  
### Step 2: Describe the problem

So I have been using these MFRC522 boards with Arduino UNO for a long time(Thx to your library :) ) There haven't been any big issues. Recently I decided to use feather LoRa for some RFID-RF projects so I bought some feather M0 LoRa and some feather 32u4 to play with. But I can't get it to work on feather M0 LoRa

- I tested each feather board with some other code to prove they all work with my IDE

- I tried different pins for SS and Reset on the feather M0, no luck

- I tried forcing hard reset in PCD_Init(), still doesn't work on the feather M0

- The exact same code and wiring(dumpinfo) will work on the 32u4 but not M0

- This is what I have
Firmware Version: 0x0 = (unknown)
WARNING: Communication failure, is the MFRC522 properly connected?

Am I missing anything here or the library has some conflict with M0 board?

ildar, 2019-04-23:
Interesting. What's the communication protocol between MCU and the RFID?
Also any reference on your board might help

ydylwj23, 2019-04-23:
Here's the MCU:  [https://www.adafruit.com/product/3178](url)   It has an ATSAMD21G18 ARM Cortex M0 processor.

The communication protocol is SPI which is the only protocol used on MFRC522 board

Rotzbua, 2019-04-23:
@ydylwj23 Can you try version `1.3.6`? There was a bug #347 which should be fixed, but maybe not..

ydylwj23, 2019-04-23:
@Rotzbua I just tried 1.3.6. It's not working. Notice that it's not the same bug: #347 is a hang bug but my bug just gives me 0x0 message.

ydylwj23, 2019-04-24:
@Rotzbua Just realized something interesting. In the header, I found the SPI clock speed is defined by a quarter of MCU's clock speed(4 MHz on 328). But since STMD chip has 3 times faster clock, a quarter would be 12 MHz which is too fast for the RFC522 chip. 

`#ifndef MFRC522_SPICLOCK
#define MFRC522_SPICLOCK SPI_CLOCK_DIV4			// MFRC522 accept upto 10MHz
#endif`

But sadly, changing the value to half(6MHz) didn't work for me.

Rotzbua, 2019-04-25:
@ydylwj23 Right, the spi clock could be the problem. Maybe some other spi specific settings are the problem. They can be found at https://github.com/miguelbalboa/rfid/blob/703a5f362e533a6cfda62dfa84538e9a6992370c/src/MFRC522.cpp#L49 .

Maybe adafruit changed something in the spi part of their toolchain. I do not see any significant change https://github.com/arduino/ArduinoCore-samd/compare/1.6.16...master , but lets give it a try.

Can you try an older toolchain? According to https://github.com/miguelbalboa/rfid/issues/347#issuecomment-364229457 I would suggest Release 1.6.16 of arduino samd https://github.com/arduino/ArduinoCore-samd/releases .

Rotzbua, 2019-04-25:
I looked into the schematics:
https://cdn-learn.adafruit.com/assets/assets/000/032/914/original/feather_schem.png?1465421956

The lora chip is connected to the normal spi interface. Maybe the lora module is active at startup and must be disabled before using the spi interface for another device.

ydylwj23, 2019-04-25:
I tried turning off the radio, it doesn't help. Some more info that might be helpful:

1. This library works with feather 32u4 LoRa. So I don't think the radio is the problem.

2. I tried Moteino M0(same MCU as feather M0), the result is even worse: mfrc522.PCD_DumpVersionToSerial() didn't print anything for me.

Rotzbua, 2019-04-26:
Did you tried the older board version of samd? (Release 1.6.16)

ydylwj23, 2019-04-26:
I don't have an Arduino samd board(Release 1.6.16). I only have a few feather M0 from Adafruit.

Rotzbua, 2019-04-26:
Sorry for my misleading description. I mean software release of the tool chain and not the hardware:

Tools->Board -> Boards Manager -> select older version

https://learn.adafruit.com/add-boards-arduino-v164/installing-boards

ydylwj23, 2019-04-26:
Yes, I was looking at board configuration versions. But only Arduino samd board has an old release 1.6.16 not Adafruit.
Current Arduino SAMD board version is 1.6.23; Adafruit SAMD board version is 1.4.1

Rotzbua, 2020-04-23:
@ydylwj23 Can I close this?

closed 2020-04-26 by Rotzbua