miguelbalboa/rfid #465: Hard-coded authentication key in `MIFARE_SetUid`

bengineer19, 2019-05-05:
Hi, 

First of all I just want to say thanks for creating an awesome open source project that I've found incredibly useful!

I noticed that in [MFRC522.cpp](https://github.com/miguelbalboa/rfid/blob/master/src/MFRC522.cpp), the value of the authentication key is hard-coded:
https://github.com/miguelbalboa/rfid/blob/a395a6c7b7ddfa729fe9bb998f5bb16f41438844/src/MFRC522.cpp#L1789-L1792

This appears to be misleading, since the [ChangeUID](https://github.com/miguelbalboa/rfid/blob/master/examples/ChangeUID/ChangeUID.ino) example appears to set the key manually, but it doesn't look like this key is actually used?
https://github.com/miguelbalboa/rfid/blob/a395a6c7b7ddfa729fe9bb998f5bb16f41438844/examples/ChangeUID/ChangeUID.ino#L45-L50

If you think this is something that could be fixed, I'd be happy to help out :) 
Cheers!

Rotzbua, 2019-05-05:
It looks like you are right, not completely sure.

lmmeng, 2019-05-06:
Maybe it is the intended behavior? 
It seems to me the new UID is "applied" to read data and written back to the card:
`// Write new UID to the data we just read, and calculate BCC byte
	byte bcc = 0;
	for (uint8_t i = 0; i < uidSize; i++) {
		block0_buffer[i] = newUid[i];
		bcc ^= newUid[i];
	}
// Write BCC byte to buffer
	block0_buffer[uidSize] = bcc;
`
And later on:
`// Write modified block 0 back to card
	status = MIFARE_Write((byte)0, block0_buffer, (byte)16);
`

lmmeng, 2019-05-10:
I see now what you mean. 
Yes, looks more like an "one shot change" (maybe it works one time with factory (not programmed) tags).

Rotzbua, 2020-07-25:
Custom key is usable on fork version.

closed 2020-07-25 by Rotzbua