miguelbalboa/rfid #477: Slave/Transponder-Mode

LexNetAT, 2019-05-29:
Hi Community,
all the time the MFRC522 acts as a master (RFID-Reader).
It is possible to use the same board as a slave (Transponder or like that) ?
For example to talk between two MCUs which uses a MFRC522 or a smart phone.

Rotzbua, 2019-05-29:
No, as I wrote in the readme: (sry, didnt documented the sources)
```
My mobile phone doesn't recognize the MFRC522 or my MFRC522 can't read data from other MFRC522

    Card simulation is not supported.
    Communication with mobile phones is not supported.
    Peer to peer communication is not supported.
```
If you find a contrary claim, post it here.

I remember the PN532 has support for your requested features.

Rotzbua, 2019-06-06:
@LexNetAT Gl with your project. If you have any update on the topic you can post it here.

closed 2019-06-06 by Rotzbua