miguelbalboa/rfid #482: question: not working on arduino uno clone

nedoskiv, 2019-06-13:
Hello,

So far I have used MFRC522 on Lua language, and Micropython, but since I'm looking for cheaper solution, decided to give it a try on atmega processor.
My goal is to wire MFRC522 directly to an atmega328 processor and use it that way. Becouse I do not want to start with wiring, decided to test it on arduino uno, I got an clone of it, but that do not work. Wired it as described, with no success, tried with another MFRC reader, same. Library is recognizing the reader but do not read tags.

I found  similar question asked in stackexchange but it was unanswered.

Rotzbua, 2019-06-13:
missing evnironment information completion required:
https://github.com/miguelbalboa/rfid/blob/master/.github/ISSUE_TEMPLATE.md

what steps done so far?
https://github.com/miguelbalboa/rfid#troubleshooting

nedoskiv, 2019-06-14:
Okay I do some testing and question is that: I use an cheap MFRC 522 from chine, with this library it was detected as conterfeit - response 0x12 and it do not detect any tags. I have tested on arduino UNO and esp8266 with 2 different MFRC readers. After that I place same reader on esp8266 with nodemcu lua and reader works without any problem. So question is that:

Is this library made on purpose to not working with readers returning 0x12 (reported as conterfeit) or that is a bug?

Rotzbua, 2019-06-14:
https://github.com/miguelbalboa/rfid#troubleshooting

Firmware Version: 0x12 = (unknown) or other random values

*    The exact reason of this behaviour is unknown.
*    Some boards need more time after PCD_Init() to be ready. As workaround add a delay(4) directly after PCD_Init() to give the PCD more time.

nedoskiv, 2019-06-14:
Well I take a look at the source, looks like there is nothing more about counterfeit, so I guess library just can't handle those readers, gonna switch to micropython or lua and use little more expensive hardware - like esp8266 instead of atmega processor. Thank you for the answers.

closed 2019-06-14 by nedoskiv

Rotzbua, 2019-06-14:
@nedoskiv thanks for your feedback. which libraries (link) did you used? what version did they reported? thanks

nedoskiv, 2019-06-15:
latest version, tried "dumbinfo" example with 3 mfrc522 readers, two of them reported 0x12 counterfeit reader and did not read tags, one of them report 0x92 and read card, bad part is that on lua all 3 readers works fine