miguelbalboa/rfid #488: Dumpinfo not working on latest versions

nedoskiv, 2019-07-17:
Hello

I was trying to make an use of MFRC522 reader on arduino uno, wired everything as described and give a shot on dumpinfo example sketch. Not working, I mean I see initialization message reported my reader as v.2 (Firmware Version: 0x92 = v2.0) and no tag is detected after that, so i give a shoot at another reader it was reported as "conterfeit" but still did not read any tag (here is the place to mention both readers work on lua/python controllers)
So I deleted the library and installed version 1.4.3 - same,
Switched to 1.3.0 and everything works fine with dumpinfo sketch. Am I doing something wrong?

nedoskiv, 2019-07-17:
ok do some more install/uninstall, On 1.4.0 not working, on 1.3.6 works fine.

Rotzbua, 2019-07-17:
Microcontroller? IDE, version? Wireing?

nedoskiv, 2019-07-17:
wiring as described in dumpinfo example, controler is arduino uno clone, IDE 1.8.8

Rotzbua, 2019-07-18:
There are only two significant changes between 1.3.6 and 1.4.4:

* reset pin code fixed
* anticollision check changed

https://github.com/miguelbalboa/rfid/compare/1.3.6...1.4.4#diff-ba13e6b0757d455f47257498e7cd156e

If you have time to test which of the two changes causes the error, it would help a lot.

Best

nedoskiv, 2019-07-19:
I'm actually not that good, to split library code and check which one of them causing the problem.

Rotzbua, 2019-07-19:
Ok, let's keep this issue, maybe some other run into the same problem and wants to fix this.

Rotzbua, 2020-04-23:
Does not seem to be a big problem. Closed.

closed 2020-04-23 by Rotzbua