miguelbalboa/rfid #489: changed authentication key, I don't know how to access the block now

dorindabassey, 2019-07-17:
hi, am a newbie but I've been read about rfid tags for sometime now, my issue is- 

i used the changekey example code to change the authentication keys of my rfid cards now i can't access the block where i changed authentication key, so i tried to run the dumpinfo example code and it shows 
[image]

dorindabassey, 2019-07-17:
uhh, I actually changed authentication key from sector 2 to sector 15

dorindabassey, 2019-07-17:
so how can I gain access to my cards with my new authentication key 

thanks in advance

nedoskiv, 2019-07-18:
you need to authenticate to block with new key, if you do not know what is the new key you write, throw chip away and start with new one :)

dorindabassey, 2019-07-18:
okay, so I know the new key, please how do I authenticate the block with the new key thanks

nedoskiv, 2019-07-18:
here is how I define my key
 ```
  MFRC522::MIFARE_Key key;
//  for (byte i = 0; i < 5; i++) key.keyByte[i] = 0xFF;
  key.keyByte[0] = 0xFF;
  key.keyByte[1] = 0xFF;
  key.keyByte[2] = 0xFF;
  key.keyByte[3] = 0xFF;
  key.keyByte[4] = 0xFF;
  key.keyByte[5] = 0x77;

```
if you cannot understand that part, should move your question to arduino forums for complete solution

dorindabassey, 2019-07-18:
okay thanks, well, I made some changes like this

void loop() {

  // Prepare key - all keys are set to FFFFFFFFFFFFh at chip delivery from the factory.
  MFRC522::MIFARE_Key key;
  for (byte i = 0; i < 6; i++) key.keyByte[i] = 0xCC;//my new authentication key

  //some variables we need
  byte block;
  byte len;
  MFRC522::StatusCode status;

  //-------------------------------------------

 and when I tried to execute the readpersonaldatafromtag example I edited, it still showed authentication failed

dorindabassey, 2019-07-18:
*my new authentication keys were set to 0xCC

dorindabassey, 2019-07-18:
😢 thanks, but am not satisfied, I don't think the sector is broken😕,  I'll read some more though

nedoskiv, 2019-07-18:
what code you use to set new authentication key?

dorindabassey, 2019-07-18:
https://github.com/miguelbalboa/rfid/wiki/How-to-change-keys%3F

nedoskiv, 2019-07-18:
Well I never used that function, I was writed my own, but on different platform to change keys. I'm not sure if that function set properly access bits. You can play with another sector/block and experiment before throw chip on garbage, you have 14 more sectors to brick :)

meanwhile I was starting to looking for alternative library since this one cannot hangle some readers.

dorindabassey, 2019-07-18:
HI, thanks so much, 
for (byte i = 0; i < 6; i++) key.keyByte[i] = 0xCC;//my new authentication key

this did the trick, the problem was my code, I was busy trying to read block1 that had a different authentication key, but now I've changed all my sectors from 1 to 15 authentication key to 0xCC and am good to go. thanks😃.

nedoskiv, 2019-07-19:
good, now close that issue please

dorindabassey, 2019-07-19:
oww.    I thought I did thanks

closed 2019-07-19 by dorindabassey